prusa3d/Prusa-Firmware #1647: Specific print causes printer to go nuts. - Z raises, claims Temp runaway

uchobby, 2019-03-18:
The attached file is g-code from S3D for a camera mount model I designed in Fusion360.  When this is printed from SD card or USB direct the printer flips out at layer 3.  It beeps, shuts off heaters shows thermal runaway on the display.  It also raises the Z and seems to continue the printing up above the print bed.

I also passed the STL through the online fixing tool just in case it was fixed there.

[CameraWallMount v1_fixed.zip](https://github.com/prusa3d/Prusa-Firmware/files/2976151/CameraWallMount.v1_fixed.zip)

omniwired, 2019-03-20:
This sounds like a S3D issue on how it prepares the gcode.
I don't think this issue is related to firmware but to the S3D slicer and your settings.

uchobby, 2019-03-21:
I think the key here is that the printer went nuts due to something in the g-code.  And I do mean nuts.  I would agree something is likely wrong with S3D but the printer should not be affected in this way.  There seems to be a defect in both S3D and the firmware.

I will be notifying S3D as well.  It would be wise to look at the file and adjust firmware to generate some error condition, to gracefully handle whatever "bad" comes from the g-code.

If I were not near the printer when these issues occurred, the printer could have been damaged.  In one case it did crash the extruder head, even though the crash detection was on.

closed 2019-04-04 by michalxfanta