prusa3d/Prusa-Firmware #1648: min temp error when preheat

lucky13hc, 2019-03-18:
in 3.7.0 rc1 when i try to preheat it shows MINTEMP ERROR I GO BACK TO 3.5.1 AND I HAVE NO PROBLEM

SoBaldrick, 2019-03-19:
Was hoping 3.7.0 would fix this issue, my 2.5s gets min temo error when bed reaches 30 degrees.

icemansid, 2019-03-19:
I came here to report the same issue. Bed heats from 22C cold to 30C, issues min temp error. re-send print and it works fine after that.

tonysmartantzoulatos, 2019-03-19:
Check issue #1621 there is problem with bed temp changes on certain bed temps along with other problems I found so far.

icemansid, 2019-03-19:
[image]
This is my issue as other stated. From a cold start, send the print, hits 28-30C on the bed temp, throws an error. If i manually heat the bed up before starting the print, it prints just fine.

icemansid, 2019-03-19:
[image]
Here is the same print re-sent after the bed was manually warmed up.

lucky13hc, 2019-03-20:
guys i found a firm that has this bug fixed and has 7x7 check it was perfect until prusa fix this https://github.com/prusa-community-forum/Prusa-Firmware/releases?fbclid=IwAR3JWitpEmNpKAjaXragFbaltJ7KYYIzaCC2iOdcGcO9P-XRVvdpn_N9noI
you must scroll down until **find 7x7 mesh bed leveling, preheat error fix, hex files**

Nippylicious, 2019-03-23:
Same Problem here since i upgraded from mk2s to mk2.5(s)

My workaround:
- Preheat to 25°c
- when 25°c is reached, hear to 30°c
- when 30°c is reached then hear to your final temperature (60°c)

I think it need to allow longer heat time before give out an error.
On mk2s no error.

bynnek, 2019-03-27:
Same problem for me on 3.7,0 and also on 3.5.2.  3.5.1 did not have this issue.  This is on a MK2.5 with MMU2.

tonysmartantzoulatos, 2019-03-27:
Check also to see if you start a print with 80c bed and ask to go 85 or 90 on second layer if you will have a problem also there. I have found this to be the second bed temp related issue of 3.6 and 3.7

cimoalpacino, 2019-03-27:
Could anyone of you try ti disable timer02, recompile the code and try again?

Kolous, 2019-03-28:
This isssue happens to me too on MK2.5 since FW 3.6 (or 3.5.2). I usually preheat and get the card to load the gcode one. The temperature inceases slightly then I get this error. After restart the preheat works fine.

icemansid, 2019-03-30:
I found a simple work-around. In your start g-code, set the bed temp to 25, 35, 45 and then start bed temp. This allows it to print without triggering the min temp error.

icemansid, 2019-04-01:
[image]
Here is my work-around. I added  the two lines below after the M104 command to set the extruder temp but before the M140 setting the first layer bed temp. This is added in "Start G-code" section.
M190 S30;
M190 S50;

OopAck, 2019-04-02:
I have tried to reflash the firmware but to 3.6 and 3.5.1; still getting the error.
Also tried the M190 Sxx lines in front M140 lines and also failed.

clintonefisher, 2019-04-05:
3.7.0 and I get the MINTEMP BED error when preheating from cold and it hits about 27C  It's not always the same temperature.  FWIW, once this error is cleared, the bed temp fell back to 17C and when preheated again did not trigger the mintemp error.

MK2.5S+MMU2S (MK2S --> MK2.5 --> MK2.5S+MMU2S)

I wonder how many others are affected and I wonder if not that many people, then could it be some leftover EEPROM values or something from an older firmware?
With the MMU2S, I upgraded my firmware to 3.6 from an older 3.1 or 3.2 version, which threw the MINTEMP BED error, and then 3.7 threw the error as well.

Don't know about others, but seems to me to be a moderate-to-high-severity issue if a lot of folks are affected.   In the meantime, I'll try the workaround and see how it works work.

lucky13hc, 2019-04-06:
https://github.com/prusa-community-forum/Prusa-Firmware/releases?fbclid=IwAR3JWitpEmNpKAjaXragFbaltJ7KYYIzaCC2iOdcGcO9P-XRVvdpn_N9noI

scroll down here to 7x7 this guy solved the preheat at 3.5.1 i tested it

SoBaldrick, 2019-05-16:
Just upgraded to 3.7.1 and issue still present on my Mk2.5S

Kolous, 2019-05-19:
This might be related to bad contact between HB connector and Rambo board. Mine was half melted and therefore it took bit more time to heat up. This is flawed by design. So when started to smell bad I soldered the wires directly to board (because this was already third connector burnt). And now the HB heats up before the mint temp error.

icemansid, 2019-05-20:
While that is a problem with the MK2 Rambo board, I have mitigated that
issue by using mosfet power boards. This allows for a very low amp draw
through the Rambo board but high amp draw directly from the power supply.
I still have the issue described in this thread. In a previous comment, I
tested this identical gcode on different firmware revisions.

ssvenn, 2019-11-10:
The problem seems to be caused mainly by cold rooms, my MK2.5S needs the stepped preheating workaround in the winter but is fine in the summer. There is a timeout where the firmware checks if the bed is still below 30C  after 50 second and if it starts at less than 20C it rarely reaches that temperature in time. The printer is in an enclosure so once it gets going it has no trouble maintaining the correct temperature.

There should probably be some new logic here where the MIN_TEMP timer saves the current sensor reading when it starts heating and then checks that the temperature is increasing compared to that reference instead of the constant of 30C

https://github.com/prusa3d/Prusa-Firmware/blob/46c74f84b6c09b6cc6863476fae21f44d34b037c/Firmware/temperature.cpp#L2107 is the current code

closed 2023-09-21 by github-actions[bot]