prusa3d/Prusa-Firmware #1651: [3.7.0-RC] Repeated enforced Z calibration during bed leveling

sgoll, 2019-03-19:
Since 3.7.0-RC I am using the new 7x7 mesh bed leveling (with 5 measurements per position). Now I have the problem that intermittently the bed leveling done before each print detects some problem, runs the enforced Z calibration and then re-runs the bed leveling.

Sometimes, this Z calibration is enforced a second or third time before finally the bed leveling succeeds.

This also happens when nothing at all was changed: the printer was switched on all the time (so no skip in the Z steppers occurred) and the steel sheet was not replaced (so it should still be aligned just like in the first run of the mesh bed leveling).

When is the Z calibration enforced? Is there some setting to tune down the sensitivity?

port513, 2019-03-19:
Have this too

JohnnyPhantom27, 2019-03-20:
Working on finding a cause. My buddy also has the MK3 printer with the new 3.7.0-RC1, his works fine. I decided to try the calibration wizard. It told me my motor belt was loose. I made it tighter and ran the test again, everything went great until the bed leveling. I did the 4 calibration points with paper, then put on the steel sheet as instructed, it got to the 2nd point in the test (front center of the bed) and failed. Same behavior as when I just tried printing. Got a failed: check nozzle and bed for debris. I cleaned the steel sheet, the bed, and the nozzle again. I also gave the screws holding the bed in place a snug turn with the hex key. re-trying...

JohnnyPhantom27, 2019-03-20:
Cleaned the bed, nozzle, leveling sensor, and gave the bed screws a snugging. I get past the wizard now, but during the first layer calibration, it fails during the bed leveling, front center each time. I hope my trial and errors helps.

PavelSindler, 2019-03-20:
Thank you all for your reports.

JohnnyPhantom27, 2019-03-20:
Figured maybe a factory reset might help, since I did have some manual leveling adjustments in the printer before going to 3.7.0-RC1, it did not :(

port513, 2019-03-20:
> Figured maybe a factory reset might help, since I did have some manual leveling adjustments in the printer before going to 3.7.0-RC1, it did not :(

I also did a factory restore with no solution on my MK3S

geeksville, 2019-03-21:
btw - From looking at the code, I don't think ya'll should bother with factory resets etc... to try and find a workaround.  It just looks like it gets confused part way through moving through the 7x7 mesh and thinks it is done (when it isn't - on mine it often happens after just the fifthish point).  I didn't look more carefully.

Shinomorisama, 2019-03-22:
I have this problem too on my MK3.   It does the leveling fine for the test,but when starting the print it actually seems faster than the test. It took 4 tries to get through the 7x7.

PavelSindler, 2019-03-22:
Did it always happen when 7x7 mesh was turned on? Or did somebody encounter this issue also with 3x3 mesh?

PavelSindler, 2019-03-22:
Also can you please send me serial log from this mesh bed leveling fail? (You can use Octoprint or Pronterface or using PuTTY for example (115200 Baud, CR & NL).

Kolous, 2019-03-22:
Same issue with MK 2.5. First 5 prints the 7x7 leveling was always sucesfull now it takes more than 10 attempts to start printing. Always failing on different point....espacially frustrating is faling on last point.

geeksville, 2019-03-22:
@PavelSindler I'm happy to send you a log if you still need it.  It always happens in 7x7 not 3x3, and it can occur on any point (sometimes 5th, sometimes not at all, sometimes last point).

Ghenghis, 2019-03-22:
only on 7x7 at different points tried 7 times last night had to switch back to 3x3 to print something. 

Also not sure why it dont probe far right side at all. it does probe far left no problem but alot of the right side isn't being probed - almost a whole square-right-side-whole-row on the flex sheet on far right side of bed, maybe this is normal but thats alot of area not probed on the whole bed.

Shinomorisama, 2019-03-22:
Not sure how to get the serial log.  In the terminal in octoprint it says "Bed leveling failed. Sensor didn't trigger. Debris on nozzle? Waiting for reset".  I've had one failure now on 3x3 on point 8. On 7x7 it fails almost every time but it is very random where. Usually in the first 2 rows. Watching the probe. The light seems to flicker 3 times on the points where it is failing just like every other point.  The calibration never has any trouble.  Tried calibration cold then G80 in terminal with cold bed. Still some failures.  Tried calibration hot and running a print still failures.  G28 it seems to work but it is moving slower.

halfluck, 2019-03-23:
Phew! glad its not just me thought i had a dodgy pinda, glad its only a FW issue :)
mine does it on 3x3 and 7x7

Kolous, 2019-03-23:
The issues I described are with 7x7 and 5 checks. When decreasing to 4 it works reliably.

Nippylicious, 2019-03-23:
Glad to hear that you guys had the same issues.
Mine 2.5s only works with 3x3.
7x7 occurs in the First (Front) row in an error.

kripobs, 2019-03-23:
a simple G28 works for me if everything is cold. also the xyz calibration was ok. but as soon as i start print and everything is heated up it stops in the middle of the front and want to force calibrate z. in both modes 3x3 and 7x7
reverting back to an older firmware so no logs from me. sorry (i also dont know how to make one :D)

sgoll, 2019-03-23:
> Also can you please send me serial log from this mesh bed leveling fail?

@PavelSindler Here is the serial output when the bed leveling fails:

```
tmc2130_goto_step 1 1 2 1000 
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
echo:busy: processing
(…)
echo:busy: processing
Bed leveling failed. Sensor didnt trigger. Debris on nozzle? Waiting for reset.echo:busy: processing
echo:busy: processing
echo:busy: paused for user
(…)
echo:busy: paused for user
tmc2130_home_enter(axes_mask=0x04)
```

sgoll, 2019-03-24:
This may have to do something with the PINDA probe temperature. For me, the problem is much more severe when trying to print PETG (230/75) compared to PLA (215/60).

cosmith71, 2019-03-24:
3x3 works _most_ of the time on my MK3.  7x7 almost always fails at least once. On my MK2.5S 7x7 is unusable and 3x3 fails so often I had to roll back to 3.6 just to get a usable machine.

PavelSindler, 2019-03-24:
Thanks for help.

Nippylicious, 2019-03-25:
do you guys have this behaivour?

1. first the bed touches smoothly the endstop
2. second the extruder toches smoothly the endstop
-- so point 1&2 are normal --
3. the extruder moves a bit to the right side X Axis
4. and then it "chrashes" into the X endstop before it start the mesh

i took a video and packed it into a zip
[Extruder_Endstop_Crash.zip](https://github.com/prusa3d/Prusa-Firmware/files/3002219/Extruder_Endstop_Crash.zip)

leptun, 2019-03-25:
Yes, I also have this problem. I think this is caused by the old  X Y home offset calculated during XYZ calibration. Maybe running it again with the latest firmware fixes the issue. I didn’t have time to test this myself, but will do so when I go to the printer next.

sgoll, 2019-03-25:
@Nippylicious @leptun I do not have this issue: the print head moves to the left-most 7x7 bed mesh point but does not crash into the X axis holder (although it comes rather close, < 1 mm I guess).

What are your XYZ cal. details at `Support` → `XYZ cal. details`? I have the following:

```
Y distance from min:
Left: 11.41mm
Right: 11.51mm

Measured skew: 0.00°

[0;0] point offset
X 0.27 mm
Y 0.91 mm
```

For the record, I was able to run the 7x7 mesh calibration fine for the last two PETG prints I did. Here's what was different:

- I flipped the flexible PEI steel sheet to its back.
- I started the prints on a cold printer (so only preheating in the start G-code).

I don't know if this may be relevant.

PavelSindler, 2019-03-25:
Did you guys run Wizard or XYZ or Z-calibration after upgrading to 3.7.0-RC1? If you didn't, please run Z-calibration from menu "Calibration->Calibrate Z". 
Crashes are caused by mesh which was moved too much to the left side. There is another issue on that. I would like to solve just Z-levleing here in this thread. Thanks.

sgoll, 2019-03-25:
@PavelSindler I did run the Z calibration (from the `Calibration` menu) repeatedly. It did not have any effect on the problem with the mesh bed leveling.

Nippylicious, 2019-03-25:
@sgoll:
my XYZ Calibratzion:

Y distance from min:
Left: 8.31mm
Right: 7.45mm

Measured skew: 0.19°
Slight skew: 0.12°
Serve skew: 0.25°

[0;0] point offset
X -5.02mm
Y -0.14mm

------------

@PavelSindler:
No, i did not run the Wizard after upgradting from 3.6.0 -> 3.7.0 RC1
I thought i was not nessecary (my bad).

When my actual print is done, i flash it back to RC1 and run the Wizard.
Then i give you feedback.

Nippylicious, 2019-03-25:
@PavelSindler:
so.. print is finished and i flashed back to RC1
Run the Wizard.

The "crash in X endstop" is still the same.
Then it run a whole 7x7 calibration before Z-adjustment (with no problems).

When it finished the Z-adjustment, i ran a mesh bed leveling from the calibration menu.
Then it crashes again against the X endstop, and aborts after point 4 with an error (Please rotate the Z Axis against frame....), i did it... then the same procedure again (crash, point 4, error........)

leptun, 2019-03-25:
In my case it might be because of the fat cable bundle hitting the frame

Nippylicious, 2019-03-25:
> 
> Crashes are caused by mesh which was moved too much to the left side. There is another issue on that. I would like to solve just Z-levleing here in this thread. Thanks.

I created a new Issue
[https://github.com/prusa3d/Prusa-Firmware/issues/1674](url)

cosmith71, 2019-03-26:
> Did you guys run Wizard or XYZ or Z-calibration after upgrading to 3.7.0-RC1? If you didn't, please run Z-calibration from menu "Calibration->Calibrate Z".

@PavelSindler 

I did the Z-calibration.  Now my MK2.5S is completing the 7x7 reliably.  No change on the MK3.  It won't complete a 7x7 without the "Some problem encountered, Z-leveling enforced..." error.  I did an XYZ and a Z calibration on the MK3.

PavelSindler, 2019-03-26:
Here is fw (just MK3 and MK3S versions) with more information written to serial log, which should help us to debug this issue:
[fw-test.zip](https://github.com/prusa3d/Prusa-Firmware/files/3008061/fw-test.zip)
@sgoll would it be possible for you to flash it to your printer and send me information from log again?

sgoll, 2019-03-26:
@PavelSindler Sure. Let me finish my current print job (12 hours to go) and I'll be able to send you the log output probably tomorrow.

geeksville, 2019-03-26:
Pavel if you can point me at a branch/commit with that extra debug info I
don't mind collecting the life on my printer (alas I need source because I
run a branch with a small change to work with my JLTX indirect filament
sensor).

(Sent from a phone - please ignore typos)

PavelSindler, 2019-03-26:
It is "7x7_polishing" branch ;-)

wavexx, 2019-03-26:
Did the "skew" limits change from 3.6.2 to 3.7.0-rc1? I wanted to test 7x7 leveling, but curiously when running the XYZ calibration the procedure now detects a slight skew, whereas 3.6.2 (and all previous) were just saying that the axes are perpendicular.

wavexx, 2019-03-26:
Just out of curiosity, just by rerunning the XYZ cal I went from 0.12 of skew to 0.06 (mmh).

PavelSindler, 2019-03-26:
No, it didn't change. You may be somewhere in between slight skew and perpendicular.

3d-gussner, 2019-03-26:
@PavelSindler Just a thought, would it be easier if we had debug (read only) options by default?
I was thinking of a kind of gcode commands which write to serial or SD card (often already implemented):
- in the past on the MK2 i used quite often M45 V and G80 V verbosity levels to troubleshoot things.
- also having the (reading not writing) dcodes active by default would make a bug report much easier. Remember TheZeroBeast found temp issues in the PINDAv2 like that and he could post an issue report with really good data.
- 'normal' users don't know how to active the debug mode and compile firmware. They would be then able to provide needed data much easier.

As every machine can differ from what you have by hand, users with specific issues could dump these information in the issue report.

Gina from Octoprint uses it all the time to analyse issues. I know that the Einsy resources are hitting the limits quite often and cannot be compared to boards like the Rpi with 'unlimited' resources. But I already wrote a proposal how-to get 8-10kb free in another topic.

PavelSindler, 2019-03-26:
Yes it is good point. We undertand it and logging is on our todo lists already. Verbosity levels in calibrations are currently disabled to save flash unfortunately.

geeksville, 2019-03-26:
Hi @PavelSindler here you go.

Test conditions:
- confirmed plate was clear of debris
- default 7x7 and three readings per point
- your branch
- the Prusa mesh bed leveling plugin in octoprint shows my bed as super level (.15mm max deviation)

Here's the relevant log output of a failed session:
```
Send: N9 M104 S160*107
Recv: ok
Send: N10 M190 S80*97
[...]
Recv: ok
Send: N11 G28 W*84
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv:   0 step=14 mscnt= 226
Recv: tmc2130_goto_step 0 13 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:   0 step=50 mscnt= 800
Recv: tmc2130_goto_step 1 49 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Send: N12 G80*34
Recv: [1.000000;1.000000]
Recv: after clamping: [1.455118;0.999955]
Recv: Min. Z: -10.000000
Recv: init Z = 0.360000, min_z = -10.000000
Recv: PINDA triggered at 0.162500
Recv: init Z = 0.362500, min_z = -10.000000
Recv: PINDA triggered at 0.150000
Recv: init Z = 0.350000, min_z = -10.000000
Recv: PINDA triggered at 0.160000
Recv: [35.000000;1.000000]
Recv: after clamping: [35.000000;1.000000]
Recv: Min. Z: -10.000000
Recv: init Z = 0.355000, min_z = -10.000000
Recv: PINDA triggered at 0.142500
Recv: init Z = 0.342500, min_z = -10.000000
Recv: PINDA triggered at 0.145000
Recv: init Z = 0.345000, min_z = -10.000000
Recv: PINDA triggered at 0.135000
Recv: [69.000000;1.000000]
Recv: after clamping: [69.000000;1.000000]
Recv: echo:busy: processing
Recv: Min. Z: -10.000000
Recv: init Z = 0.320000, min_z = -10.000000
Recv: PINDA triggered at 0.100000
Recv: init Z = 0.300000, min_z = -10.000000
Recv: PINDA triggered at 0.090000
Recv: init Z = 0.290000, min_z = -10.000000
Recv: PINDA triggered at 0.125000
Recv: [3;0]: Z_offset = -72
Recv: Bed leveling, point: 3, calibration Z stored in eeprom: -72, calibration z: -0.562500
Recv: [103.000000;1.000000]
Recv: after clamping: [103.000000;1.000000]
Recv: Min. Z: -1.562500
Recv: init Z = 0.327500, min_z = -1.562500
Recv: PINDA triggered at 0.092500
Recv: init Z = 0.292500, min_z = -1.562500
Recv: PINDA triggered at 0.130000
Recv: init Z = 0.330000, min_z = -1.562500
Recv: PINDA triggered at 0.097500
[... 607 lines cut ...]
Recv: ok
```

Here's the gcode I used (makes it easier to retest rather than printing a model):
```
M201 X1000 Y1000 Z1000 E9000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X8 Y8 Z0.4 E1.5 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
G90 ; use absolute positions (for octolapse) kevinh
M83  ; extruder relative mode
M104 S160 ; kevinh preheat extruder partially - don't wait - but a low temp to prevent drooling
M190 S80 ; set and wait for bed temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
@BEDLEVELVISUALIZER
G81 ; kevinh send mesh level results to octoprint plugin

M107 ; turn off fan
G1 Z50 X0 Y200; home X axis
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M84 ; disable motors
```

cosmith71, 2019-03-27:
> @PavelSindler
>
> I did the Z-calibration. Now my MK2.5S is completing the 7x7 reliably. No change on the MK3. It won't complete a 7x7 without the "Some problem encountered, Z-leveling enforced..." [...]

Spoke too soon.  It won't complete a 7x7 or a 3x3 this morning.  Re-doing XYZ calibration to see if that helps.

sgoll, 2019-03-27:
@PavelSindler I managed to flash the modified firmware and reproduced the error (the last few prints the error did not happen as often as previously, so it took me some attempts but then it happened again).

I am attaching two runs: one that finished mesh bed leveling without problems and one that failed (with an error of "big deviation").

- [prusa_good.txt](https://github.com/prusa3d/Prusa-Firmware/files/3012711/prusa_good.txt)
- [prusa_bad.txt](https://github.com/prusa3d/Prusa-Firmware/files/3012712/prusa_bad.txt)

PavelSindler, 2019-03-27:
Actually it seems now that this issue happens when printer is placed on table which resonates during mesh bed leveling Z-probing. We were able to replicate this issue on two printers. These printers were failing during mesh bed leveling with the same error message that was reported by you (high deviation between currently measured Z-coordinate and avg Z value from previous measurements). When it was moved to different table deviation significantly decreased. Do you have your printers on stable table?

geeksville, 2019-03-27:
@PavelSindler yes.  This table (the rolling wheels are locked and on a hardwood floor).  The feet I'm using are not the stock prusa rubber feet (because those kept coming off) - rather PETG + M3 bolt with felt pads (see photo).  
[image]

geeksville, 2019-03-27:
Perhaps fix by a small delay after the move? 100ms?

PavelSindler, 2019-03-27:
You can try this fw, which I believe should solve the issue:
[PavelSindler_fw_mk3-3.5.3+7x7_polishing.2179.g58632975.201903271926.zip](https://github.com/prusa3d/Prusa-Firmware/files/3014973/PavelSindler_fw_mk3-3.5.3%2B7x7_polishing.2179.g58632975.201903271926.zip)
When this issue with high deviation occurs for the first time, measurement is repeated with old/standard Z movements. If it fails also, Z-leveling is enforced.

sgoll, 2019-03-27:
@PavelSindler Thanks, I'll try that firmware after my current print has finished.

Regarding the unstable table: Not really. I have my printer on a quite sturdy wooden bench-like table, using the Prusa-supplied rubber feet. There is no noticeable vibration on the table when the printer is moving its axes.

PavelSindler, 2019-03-28:
Ok, so this theory is probably wrong.

geeksville, 2019-03-28:
Hmm.  Could it be the pinda probe getting slightly different readings due
to a small amount of inductive current generated each time the bed heater
is toggled on/off?  Ie perhaps compare to a build that briefly disables the
bed heater before each reading?

(Sent from a phone - please ignore typos)

sgoll, 2019-03-28:
That would explain why I never seem to get this error when I manually run the 7x7 calibration from the Calibration menu but almost always when running as part of the start G-code.

geeksville, 2019-03-28:
And just musing a bit more on this theory: perhaps it is worse in 3.7 not
just because of the 7x7 levelling but the change to shorten the period on
the bed heater PID loop (which works great to solve the power supply
clicking).  More edges on that square wave == more chance of doing a probe
while an edge occurs?

geeksville, 2019-03-28:
It also might explain why people on reddit are saying it happens more often
with PETG (because the bed heater gets kicked on more often for higher bed
temps)

PavelSindler, 2019-03-29:
I don't see any obvious difference in deviation between Z-measuremets with heatbed off (target temp 0) and preheated (tried 60 C and 100C).

sgoll, 2019-03-30:
@PavelSindler I have now run five or six prints with your firmware and have not seen any issues with the mesh bed leveling. Twice the leveling stumbled and took longer to complete but that would be your workaround, I guess.

The only thing that's somewhat annoying is that the firmware keeps reminding me at the start of every print that I should upgrade to version 3.6.0 but I think I can get rid of that in the start G-code.

cimoalpacino, 2019-03-30:
If I'm not wrong, you should disable the M115 command in the start gcode sequence in Slic3r

primalmotion, 2019-03-30:
to me this has been happening since 3.5.something. Getting z leveling pass is just luck. always fails on point 6 or 7

PavelSindler, 2019-04-01:
Yes, I think it is ready for release. Our testers wanted to do more tests on mesh bed leveling today, but it seems ok and hopefully will be released today.

3d-gussner, 2019-04-01:
@PavelSindler don't release anything on April 1st ... People may think it is an April fool :-)

[51 more replies]

closed 2020-07-13 by JohnnyDeer