prusa3d/Prusa-Firmware #1654: Calibration order change

mirekhk, 2019-03-19:
Hi all,

I wonder if it is possible to change the calibration order. The printer MK3 is checking 9 points before any print. Let's name the order 1,2,3...9.  The 5th check point is in the middle of printing area and it is exact point where the rest of filament is dropping down. If we change the order of check points to 1,2,3,4,9,8,7,6,5 the drop point will be the most right corner of the print bed. If there is a dropped piece of filament it will not damage the print.

I am not sure if the order of checks is not fixed for some reason. Anyway it would be nice if it could be changed.

thx
mirek

PavelSindler, 2019-03-20:
It is indeed possible. But I think that filament can drop on any point (depending on which material it is, what is the temperature etc.).

mirekhk, 2019-03-20:
Yes, that's true, but my experience says it drops down in the middle:-)
I am not complaining and not pushing for it. It was just an idea for implementation.
thx even for letting me know that it is possible.
mire

GurliGebis, 2019-03-21:
It is better to change the startup gcode, so it does not heat the nozzle up too much before calibrating the bed.

rtyr, 2019-03-21:
We have tried this approach and decided to not implement it for now. Lowering temperature of the nozzle (if LCD preheat is used) and heating up again adds significant time to the print start (it is about 2 minutes for 215C/160C/215C scenario). There might be also an issue with filament compromising mesh bed leveling when temperature of the nozzle is not high enough to "squish" it.

3d-gussner, 2019-03-22:
I have a start gcode
- where the hotend will be heated until it reaches 'printtemp-30°C' 
- set hotend temp to 'printtemp' BUT not waiting to reach it
- start mesh bed leveling
- wait hotend temp reaching 'printtemp'
- start purge line 

The missing 30°C will be reached at the end of the meshbed leveling and so i don't have drops allover

wavexx, 2019-03-22:
I implemented something similar on another printer (custom i3). I lower the nozzle by 5c and turn on the blower while probing (in this specific case, a bl-touch).

There will be a minimal amount of ooze (<1mm) but it's kept in check by the blower fan making it harder. Also, since it's being blowed on, it won't stick as easily to the bed.

At the last calibration point I set the final nozzle temp and turn off the blower. For PLA, going from 205 to 210 it only takes a few seconds, so by the time the head is at 0,0 again the temp is halfway there.

I never bothered to test it on the MK3 yet, mostly because the temperature delta needs to be tuned per-material/brand if you want to keep it fast.

michalxfanta, 2019-04-04:
When testing 7x7 mesh bed leveling, we found that preheating the nozzle and the PINDA probe improves calibration results. So dripping a small amount of material is a small price for a better first layer.

closed 2019-04-04 by michalxfanta