prusa3d/Prusa-Firmware #1655: [3.7.0-RC1] MK3 x-carrier: Z-axis calibration fails with MK3S firmware

Spacemarine2018, 2019-03-19:
I'm using the bondtech extruder together with the MK3S Sensor from the MK3S Upgrade. In order to make use of the MK3S Sensor, the MK3S firmware needs to be installed. Since Bondtech extruder is only compatible with the MK3-style X-axis carrier back-cover, the resulting Z-Axis length is the old MK3 z-axis length. MK3S firmware expects this axis to be longer and fails therefore on calibration.

A lot of users with Bondtech extruder are switching to MK3S Sensor and are experiencing this problem.

Possible solution:

- Introduce an "I know what I am doing, ignore and continue" option in the wizard and other calibration routines if a failure occurs.

- Provide a firmware with the option to choose between MK3 and MK3S Sensor type

- Provide a firmware for MK3 with MK3S sensor hardcoded

- Increase the range of accepted z-axis lengths to accept both MK3 and MK3S type lengths.

Spacemarine2018, 2019-03-20:
Bertrand Giot released a modified version for the MK3 with the MK3S sensor which seems to work for me:
https://github.com/bgiot/Prusa-Firmware-BMG/releases/tag/3.7.0rc1

PavelSindler, 2019-03-20:
We are busy with supporting our hardware and it is not possible for us to support all possible modifications. You can however modify function calibrate_z_auto(), to achieve what you need.

closed 2019-04-05 by michalxfanta