prusa3d/Prusa-Firmware #1659: Missing/disappearing preset profiles for print settings

haleypearse, 2019-03-19:
**Version**

slic3r PE 1.42.0-alpha5+
Mac OS 10.14.2 Mojave

**Behavior**

All of the default and custom presets for print settings have disappeared, regardless of preferences checkboxes such as "suppress default presets" or "show incompatible ... presets." This requires me to redo all print settings every time I open slic3r. The filament and printer settings presets are normal. If I save a preset, it will show in the drop down menu, but none others. I found the preset .ini files in the application support folder, but I can't get them into slic3r. 

[image]

sgoll, 2019-03-20:
Looks like this should be posted in https://github.com/prusa3d/Slic3r/issues. Also be aware that there are different support folders for different Slic3r branches (alpha, beta, release versions).

PavelSindler, 2019-03-20:
Yes, please post it to Slic3r repository and close it here. Thank you.

haleypearse, 2019-03-21:
The problem is temporarily resolved by loading a project with profiles saved to it. Now I can see print profiles. But I expect the problem to recur. I can close the issue here, but I don't understand why it doesn't belong here in PE.

JohnOCFII, 2019-03-21:
> But I expect the problem to recur. I can close the issue here, but I don't understand why it doesn't belong here in PE.

Because this is the firmware repository, not the Slic3r repository.  They are separate.

haleypearse, 2019-03-21:
Oh!

closed 2019-03-21 by haleypearse