prusa3d/Prusa-Firmware #1663: Normal/Stealth mode change during levelling causes levelling failure

dkvinyls, 2019-03-20:
Levelling fails if the mode is changed from normal to stealth during 7 x 7 bed levelling. Change from stealth to normal during levelling not tested.

First time this happened the printer reported levelling failed due to debris on the nozzle, and to wait for sensor reset. This didn't happen and required power to be cycled. Whilst trying to reproduce the condition the printer reported "some problem on Z" or similar words and proceeded to raise the z to max height and re-home before starting the levelling process again.

I know this won't be a common use scenario, but I had forgotten to change the mode before hitting print.

dkvinyls, 2019-03-20:
I haven't tested this with the 3 x 3 levelling, but I thought it was pertinent to mention which option was in use.

michalxfanta, 2019-05-16:
Yes, changing from normal mode to stealth mode cause problem at mesh bed leveling, but after evoked z-leveling everything continues correctly.

closed 2019-05-16 by michalxfanta