prusa3d/Prusa-Firmware #1669: [Feature Request] Add delay timer

qqgg231, 2019-03-23:
Please allow to start job from sd card with chosen delay, for example after 2 hours from the time user chose job

It should be very easy to add this feature

Neolker, 2019-03-25:
Not sure if this is implemented, but you should try add at the beginning of the .gcode file:
`G4 S7200 ; Wait for 2 hours`
See here: http://marlinfw.org/docs/gcode/G004.html

3d-gussner, 2019-03-26:
@Neolker That will work, but if you have a gcode that you reprint often/ or later it will wait always 2 hours before it starts.
@qqgg231 Don't know how many users need/will use that. Maybe Octprint could help you to schedule prints.

qqgg231, 2019-03-30:
@3d-gussner 

Now I see it is still not implemented in OctoPrint:
https://github.com/foosel/OctoPrint/issues/349

michalxfanta, 2019-04-02:
Delaying the start of the print is against the recommendation to check the correct printing of the first layer.

closed 2019-04-05 by michalxfanta