prusa3d/Prusa-Firmware #1673: Part of second layer missing

leftspin2k, 2019-03-24:
Originally posted in the Slic3r section, they say the gcode looks fine.
https://github.com/prusa3d/Slic3r/issues/2007

Version

firmware 3.5.1 release build
Prusa MK3 stock

Slicer
1.42.0-beta_win64
Operating system type + version

Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 17134), 64-bit edition
3D printer brand / version + firmware version (if known)

Behavior

Slice using the stock configuration for 0.3mm Draft mode

export

Print the file.

first layer prints fine, second layer does not extrude during first 20% of the layer. printer moves over the region.

Print 1 and print 2 top sides

Reproduction of the print defect.

[image]

Print 1 top side and Print 2 bottom side.

Shows that extrusion picks up at the same location in both prints. Not a filament feed issue unique to 1 machine.

[image]

Project File (.3MF) where problem occurs

https://github.com/prusa3d/Slic3r/files/2998348/Draft.bug.zip

inquisitiveKangaroo, 2019-03-26:
Hi,
did you try to flash a newer version of firmware?
Kangaroo

michalxfanta, 2019-04-02:
Hello, the first layer is printed with 215 °C that's ok, but the second layer is switched to 195 °C and it is too low for PLA.

michalxfanta, 2019-04-02:
196/5000
During temperature change, the temperature may drop temporarily below 195 ° C and the nozzle may temporarily clog. If the temperature drops below 170 ° C, the extrusion stops completely until the temperature rises again.

leftspin2k, 2019-04-04:
Thanks for that explanation! It would be nice if there was a warning for that in Slic3r.

closed 2019-04-04 by leftspin2k