prusa3d/Prusa-Firmware #1674: [3.7.0-RC1] Extruder crash into X Endstop before Mesh Leveling

Nippylicious, 2019-03-25:
do you guys have this behaivour?

1. first the Y Axis touches smoothly the Endstop
2. second X Axis (Extruder) toches smoothly the Endstop
3. the extruder moves up a bit and to the right side X Axis
-- so point 1 to 3 are normal --

4. then the Extruder "chrashes" into the X endstop before it start the mesh leveling

I took a video and packed it into a zip
[Extruder_Endstop_Crash.zip](https://github.com/prusa3d/Prusa-Firmware/files/3002219/Extruder_Endstop_Crash.zip)

------
_Originally posted by @Nippylicious in https://github.com/prusa3d/Prusa-Firmware/issues/1651#issuecomment-476100870_

mgg4, 2019-03-25:
I had issues (also reported), with the crash happening before the first point (front left) is checked.  It proceeded to do all 9 (or 49) points, and then reported the crash.  The mesh bed leveling process repeated, but this time without the crash.

I had occasion to tear down my hot-end to clear a nasty clog.  Upon rebuild, I ran through the full Wizard to test and recalibrate everything.  Since that time I have not had any further issue with this crashing.  From this, I surmise that something about the 3.7.0 firmware is using (or misusing) a calibration value which is interpreted differently than previous versions of the firmware.

In any case, you might try doing a full calibration using the Wizard as a workaround to see if this helps you.

Kolous, 2019-03-25:
I had this issue when upgraded to 3.7.0RC1. Had to force push the x carrige left and then I glued a 1mm thick stripe on the x carrige to trigger X enddstop.

Kolous, 2019-03-25:
Also I rearanged the cabeling to be sure (there is not enough space) but this wasn§t the cause of the issue. After 10 attempts I removed the stripe and it works now. This convinced me that it is more firmware than hardware issue.

Nippylicious, 2019-03-25:
i think this is definitely a firmware issue, because the first time it touches the endstop gently and the second time it ram / crash into.

Kolous, 2019-03-25:
Mine was crashing even the first time.

cyberboom2, 2019-03-26:
My mk3 crash same (#1645)

PavelSindler, 2019-03-26:
What happens if you home axes (calibration->autohome) and then go to "settings -> move X" and go to minimum X coordinate. Does it crash? Is it possible to go to X = 0?

Nippylicious, 2019-03-26:
When i "home" it, then the extruder touches gently the endstop (as it always was)

The crash is after home before mesh Leveling.

In 3.6.0 there is Not such a behaviour

ZachHenkel, 2019-03-26:
Also present on MK3

Alandovos, 2019-03-26:
Also seeing this in MK3

PavelSindler, 2019-03-26:
I have found MK2.5S printer on which I can see this behaviour finally!

PavelSindler, 2019-03-26:
Ok, it has been fixed now. Thanks for help.

cyberboom2, 2019-03-26:
Are ots been fixed on mk3

MarcoZ76, 2019-03-27:
> 
> 
> Ok, it has been fixed now. Thanks for help.

When will you correct the firmware?

dikodahan, 2019-03-30:
I have just experienced it as well. Any time frame estimate for the release of an updated firmware for this issue?

cyberboom2, 2019-03-30:
Its crashing anyway

cyberboom2, 2019-03-30:
> Ok, it has been fixed now. Thanks for help.

its crashing anyway

cyberboom2, 2019-03-30:
Its minimum coardinate is +003.2  MK3

cyberboom2, 2019-03-30:
And its crashing

cyberboom2, 2019-04-02:
Well done Pavel its not crashing anymore , works fine 7x7

closed 2019-04-05 by michalxfanta