prusa3d/Prusa-Firmware #1679: 3.7.0-RC1 - Print stops during first layer

dikodahan, 2019-03-30:
I just got my upgrade kit to MK3S and MMU2S. After completing the upgrade to the printer and upgrading to 3.7.0-RC1, my prints kept failing consistently during first layer print. Sometimes in the middle of printing it and sometimes right after finishing first layer print.
Since I also encountered another issue with this FW level (https://github.com/prusa3d/Prusa-Firmware/issues/1674) I decided to try to downgrade FW to the official 3.6.0. Prints have no problem printing whatsoever. This is obviously an issue with the new 3.7.0-RC1 code level.

michalxfanta, 2019-04-02:
Hello, could you explain exactly how the print failed? Is it happening with FW 3.7.0 final?

dikodahan, 2019-04-02:
I think I found what the issue is.
After the HW upgrade of the printer I did not also change the printer profile in Slic3r. With the final FW and the profile updated I am no longer experiencing this issue.

closed 2019-04-02 by dikodahan