prusa3d/Prusa-Firmware #1680: Self-test nozzle crash on MK3 without steel sheet

wavexx, 2019-03-31:
I did the self-test several times so far, and I re-did another one recently to test the 7x7 leveling on 3.7.0-RC1, both before and after the XYZ calibration.

During the self-test, the nozzle is lowered until some triggering point. I noticed already early on that there's quite a difference if you have the steel sheet on or not: without the steel sheet, the nozzle is brought *very* close the bed. With the steel sheet, there's significant margin left before it's stopped.

In my last self-test I didn't have the sheet and the nozzle pushed down on the head-bed for roughly 1mm. This was actually done just after a XYZ calibration. I have no reason to believe this has anything to do with 3.7.0-RC1 specifically.

I probably did the self-test more than 10 times for sure, and never had a crash so far, but I noticed there's already a report about this: #605. For reference, my live-z-adjust is around -0.9, and XYZ calibration never gave me the impression of getting *this* close.

michalxfanta, 2019-04-02:
Hello, for 7x7 mesh bed leveling, the steel sheet must be placed on the bed.

wavexx, 2019-04-02:
This is about the self-test, not the leveling.

michalxfanta, 2019-04-04:
OK, does this problem remain with the final FW 3.7.0?

wavexx, 2019-04-05:
On Thu, Apr 04 2019, Michal Fanta wrote:
> OK, does this problem remain with the final FW 3.7.0?

This was reproduced on v3.7.0-RC1, by looking at the commits to the
final 3.7.0 I don't see any change in the self-test procedure in
ultralcd that could have fixed the issue in any way.

By briefly looking at the code in lcd_selfcheck_axis, we move Z in steps
of 1mm, checking for the endstop value at each iteration.

By what I've seen, if we start to move just at the boundary before the
pinda triggers while the distance is lower than a full millimiter we
will hit the bed.

Given that Z calibration still hovers correctly without touching, we
should maybe choose a better XY position on the plate that triggers the
probe earlier or reduce the stepping towards the end.

wavexx, 2019-04-06:
See the discussion on #1703. The proposed PR fixes the issue for me and has an improved behavior overall.

wavexx, 2022-09-27:
I think I can close my own bug report ;)
Didn't have PINDA offset issues in a long time.

closed 2022-09-27 by wavexx