prusa3d/Prusa-Firmware #1681: [Idea] Disable Steppers on MMU2 when not in use?

ggrote, 2019-04-01:
It would be nice if the steppers on the MMU2 will be disabled when they are not in use while printing. Especially if a print does not need to change the filament.
Otherwise they are making a constant rustle sound.

dikodahan, 2019-04-01:
+1 for that!

3d-gussner, 2019-04-01:
Shutting down all mmu2 steppers could get a bigger issue as these could loose their position. 
- disable the mmu2 extruder shouldn't be an issue
- disable the mmu2 idler would need to revoke every time it wakes up, as the position is critical to load/unload filament
- disable the selector would be the biggest issue as it could move and make unloading very hard.

michalxfanta, 2019-04-02:
Exactly like @3d-gussner said, keeping steppers on is on purpose.

closed 2019-04-02 by michalxfanta

ggrote, 2019-04-02:
Ok makes sense. Thank you anyway for the explanation.