prusa3d/Prusa-Firmware #1682: [3.7.0-RC1] Fillament sensor issue - loading, runout detection

Kolous, 2019-04-01:
After upgrading to MK2.5S MMU2S and flashing 3.7.0-RC1 I printed succesfully multiple single color objects. Then I noted that in the FW the fillament sensor being disabled by default. After enabling the filament is not loaded properly on beginning of a print because is pushed against the bondtech wheel and never gets loaded. When kept disabled the print starts fine but nothing happens when the filament really runs out - Finda not detecting this?? I verified both sensors in the support menu and they are giving both correct response.

michalxfanta, 2019-04-04:
Does this problem remain with the final FW 3.7.0?

Kolous, 2019-04-04:
In 3.7 filament loading works fine with filament sensor enabled. I assume that runout will work too,

TimothyGold, 2019-04-05:
I think there might still be issues in the latest release of 3.7.0.  I reported an difference noticed coming from 3.6.0 in https://github.com/prusa3d/Prusa-Firmware/issues/1689 but also realized as the discussion went on there seems to be a conflict between MMU2s and load to bondtec gears and leaving the filament sensor on.  I had to turn of the sensor for filament runout to get the multimaterial print to work reliably.  Ideally we should be able to have both work in tandem but it does not seem to work.  Rather than start a new issue I thought I would add a comment here to see if this might be related to the overall issues and conflicts users are seeing.

closed 2023-09-26 by github-actions[bot]