prusa3d/Prusa-Firmware #1685: [3.7.0] FlashAir support brocken

m-cas, 2019-04-02:
If I select card type FlashAir the card will not be recognised. It will if I set it to normal.

m-cas, 2019-04-03:
Tested again, even if I switch off or reset the printer with the front X button, the card (if the setting is FlashAir) will not be recognized. Switching to normal everything seems to be working ok.

What is supposed to be different when Flashair is selected anyway?

kazibole, 2019-04-07:
For what it's worth, my FlashAir card is working on 3.7.0 on an MK3.

m-cas, 2019-04-07:
did you do anything special after the fw installation? In my case selecting in the settings “FlashAir” makes the card unreadable. Better question yet: how have you configured your FlashAir? That could be a difference...

kazibole, 2019-04-07:
I did not have to do anything special. My printer is set to FlashAir (not normal). I am not home so I can’t verify my FlashAir card config right now.

ochm, 2019-04-12:
My Toshiba FlashAair works corectly, printer is set to FlashAair

m-cas, 2019-04-12:
Interesting, is it a v4? How is it configured?

-
From my phone while mobile, please excuse any errors.

ochm, 2019-04-14:
> Interesting, is it a v4? How is it configured?

Sure it is v4. Config done exactly step by step by [guide](https://josefprusa.cz/bezdratovy-3d-tisk-snadno-a-rychle-sd-karty-toshiba-flashair/). 

Insert card to your PC/reader
On the card find and backup (just for sure) file named CONFIG, then edit them to something like:

[Vendor]
APPSSID=your-wifi-name
APPNETWORKKEY=your-wifi-password
CID=02544d535733324740e35b5979010b01
VERSION=FA9CAW3AW3.00.01
APPMODE=5
APPNAME=3D_printer
LOCK=1
PRODUCT=FlashAir
STA_RETRY_CT=0
UPDIR=/G-Code
UPLOAD=1
VENDOR=TOSHIBA
WEBDAV=2

Save the file.
Don't forget create directory used in UPDIR direction on your card.
Reinsert the card in your reader, or insert card to printer.

Check if the card is accessible by //ipaddress_of_card (you must reserve IP in your DHCP server for card connection)

m-cas, 2019-04-14:
Thanks, here my V4 config file… we are pretty much on the same page.
I need to test more…
Btw you are behind with your sw update :D

[WLANSD]
ID=MACMK3S

[Vendor]
APPSSID=XXX
APPNETWORKKEY=XXX
CID=XXX
VERSION=F15DBW3BW4.00.03
APPMODE=5
APPNAME=MACMK3S
LOCK=1
PRODUCT=FlashAir
STA_RETRY_CT=0
UPDIR=/GCode
UPLOAD=1
VENDOR=TOSHIBA
WEBDAV=2
WLANAPMODE=0x82

ochm, 2019-04-16:
Maybe I'm behind with my sw update, but my FlasAir works OK! :)

m-cas, 2019-04-16:
Very much so :)

-
From my phone while mobile, please excuse any errors.

mvasilakis, 2019-05-02:
m-cas  - I copied and pasted your config and ochm's . Seems your line endings may be an issue. Your config pastes contiguous as one line.
I don't know how this would affect the printer seeing it. Just throwing this out there.

m-cas, 2019-05-02:
mvasilakis - should just be a copy/paste issue in my posting. The config on the card was edited with “vi” on my mac with the card mounted locally. Should be a normal unix file and the config wasn’t touched after the FW change.

kyjak, 2019-05-12:
My FlashAir4 also stopped working on 3.7 and up (3.7.1 now).
It works on PC, does not get recognized either in Normal nor FlashAir mode.

m-cas, 2019-05-17:
I've upgraded to 3.7.1 and, upon printer start, I can access the card via network but my MK3S doesn't recognize the card until I pull it out and back in. Changing setting from FLASHAIR to NORMAL doesn't change the behaviour.

uniacke1, 2019-06-10:
I'm seeing this as well. I think the card takes too long to initialize or the printer is querying it too soon - if I remove the card just enough that the Mk3 thinks it has been removed (but it hasn't lost power) and re-insert it, then I can browse the card's contents (but I don't see the IP nor IP address entry in the support menu despite the Settings option being set to FlshAir)

m-cas, 2019-06-10:
The behaviour I see is:
- starting the printer the card is accessible via IP but not on the printer itself (supporting the theory of a timing issue when starting up)
- remove the card end immediately reinsert makes the card visible both via IP and lically to the printer
- changing the setting doesn‘t change the above at all
- the IP in the support page went away a couple of major FW releases ago; not a big loss for me as I use a reserved IP address for the card

-
From my phone while mobile, please excuse any errors.

uniacke1, 2019-06-10:
Yeah, item 2 is the real sticker; means there's some card jiggling involved to get it working.

kyjak, 2019-06-11:
Is there any progress regarding this item?

danbuchwald, 2019-07-20:
Any updates for this? I'm coming across the same issues.

ochm, 2019-07-20:
And what in 3.7.2? 
I'm running 16GB W04 card without any issues (from some 3.6. version), still all OK and reliable.

uniacke1, 2019-07-20:
Timing/startup still exists in 3.7.2; sometimes card has to be removed to get the printer to see it after boot.

rhounsell, 2019-08-08:
For my Toshiba 64GB W4 FlashAir (reformatted to 64Gb Fat32), I had to change Sd2Card.cpp::Init() to send some 0xFFs after sending the initial CMD0 and before sending CMD8 to give the card some extra clock cycles. Otherwise CMD8 returns R1_ILLEGAL_COMMAND. I'll be submitting a pull request shortly with my fix. Sounds like not all cards need it, and/or the timing is marginal.

Prusa Mk2.5, 3.7.2 Firmware.

m-cas, 2019-08-08:
great news! Thanks a lot!

neslekkim, 2019-08-17:
I have had problems with this also, was working before, but didn't think about firmware versions sadly, but I always needed to restart (the X button) the printer to get the card to be found by the printer. The card was on net, so that part was working.
Upgraded to 3.8.0 RC1, and now it works like before again (using FlashAir v4, 32GB)

rhounsell, 2019-08-17:
I don’t see any specific changes in the 3.8.0 RC1 release  that would directly impact card initialization.  I’m guessing that the timing for some cards can be marginal and other changes in this release have adjusted the timing enough for your card to start working.

vintagepc, 2019-08-17:
Yes, some cards are definitely marginal. My W04 works sometimes and other times it took too long and I have to jiggle it to get the printer to see its contents.It's right on the edge.

neslekkim, 2019-08-17:
I find that if I take out the card, and put it back in, then it does not read the card, then I have to reset the printer again.

leptun, 2019-08-17:
Who here experiences the issue, applied the #2088 patch and got it working reliably?

vintagepc, 2019-08-17:
> I find that if I take out the card, and put it back in, then it does not read the card, then I have to reset the printer again.

You need to remove it just enough to retain power to the card and yet trip the card present switch. Otherwise you powercycle the card and end up in the same situation.

> Who here experiences the issue, applied the #2088 patch and got it working reliably?

I'll take that patch for a spin.

kazibole, 2019-08-17:
There is also a new FlashAir W-04 software update v4.00.04 released August 1, 2019 with "bug fix" as the description. Not sure if this will help at all.

https://jp.toshiba-memory.com/ww/support/download/flashair/w04/update02.htm

neslekkim, 2019-08-17:
Thanks, after upgrading the card, it's not recognized at all in the prusa. It's on the net, so the webdav/fileshare is working, but the prusa doesn't read the card anymore.
My version was 4.00.02, my friend have same problem as my card now, and he had v4.00.03, so maybe something was introduced with the previous firmare from Toshiba.

neslekkim, 2019-08-18:
Is anyone sitting on the files for installing v4.00.02 on the FlashAir?, so I can try to downgrade it again?, can't find anything on Toshiba pages for older files.

leptun, 2019-08-18:
@neslekkim What about this? https://www.touslesdrivers.com/index.php?v_page=23&v_code=55922&v_langue=en

neslekkim, 2019-08-18:
Wow thanks! 
I just testet again this morning, put the card in with the v4.00.04 in mk3, and now it recognized the card. Strange, I couldn't get it to work at all yesterday.
But now with the 02 version as backup, I can always go back if I get more problems.

rhounsell, 2019-08-22:
> > Who here experiences the issue, applied the #2088 patch and got it working reliably?
> 
> I'll take that patch for a spin.

@vintagepc Any luck? Just curious if my fix works for anyone else.

vintagepc, 2019-08-22:
I can't say I've observed the issue since applying it (I have seen "Sending 0xFF" in the startup serial console on occasion which leads me to believe it may have prevented the issue once or twice), but in all fairness I'm also running a debug build for #2105 which could also be slowing things down a bit and giving the card more time.

leptun, 2019-08-22:
The cmdqueue debug shouldn’t influence startup since I don’t think any command is queued when you start the firmware.

rhounsell, 2019-08-22:
FWIW “Sending OxFF” will be sent on the serial line every time a card is inserted.

neslekkim, 2019-09-08:
so, it seems like flashair 4.00.04 works stable on my prusa, and also installed 3.8.0, but If I take the card out, and insert it again, it will not read, so  I need to reboot the printer. I guess there need to be a delay before reading the card, so the card manages to boot.

MatinF, 2019-11-13:
Does anyone know how to downgrade a FlashAir from 4.00.04 to 4.00.03? We've accidentally upgraded a lot of cards and they no longer work with our application - but we're unable to see any guides on how to downgrade a card. Any tips would be appreciated.

leptun, 2019-11-13:
@MatinF Do you have the 4.0.0.3 “update” exe? I think that the upgrade process can be used also for downgrading. If you don’t find the update, let me know and I’ll search for it.

MatinF, 2019-11-13:
@leptun i do have the 03 exe - but when I run it on an updated flash air it tells me that the newest version is already installed and exits.

I tried “cheating” it a bit by deleting the config from the SD card before as well as during the update process when using the 03 upgrader. I also tried writing 02 manually instead of 04 in the config version to try and mimic a 02 card. However, when the installer verifies the version it keeps trying that step with no luck.

Any suggestions?

neslekkim, 2019-11-13:
I thought I needed to downgrade, but newer did. 
After some fidling, then the 4.00.04 works fine on my printer.
What firmware do you have on your prusa?, can it be related to that?

MatinF, 2019-11-13:
We use the card also for other use cases, but figured I would try to ask here as there seem to quite a few users. Sorry if off topic; we’d really like to find a downgrade solution

m-cas, 2019-11-13:
for what is worth, the combination of FW 3.8.0/3.8.1 and 4.00.04 works pretty nicely for me

ronakjindal10, 2020-03-06:
> We use the card also for other use cases, but figured I would try to ask here as there seem to quite a few users. Sorry if off topic; we’d really like to find a downgrade solution

Hey Martin, did you find a way to downgrade the firmware. My app isn't working properly because of the changes they made.

MatinF, 2020-03-06:
No we never found a way, we updated our firmware to support the new version

ronakjindal10, 2020-03-07:
What changes did you make to your firmware to support the new firmware? I'm not able to find a solution to this problem. Thank you

MatinF, 2020-03-07:
We use the cards in another product that we develop ourselves

leptun, 2020-03-07:
You can read about compatibility changes for newer fw here: https://github.com/prusa3d/Prusa-Firmware/pull/2397

ronakjindal10, 2020-03-07:
That's great MatinF. I am working on something different myself. The transfer speed is too slow sometimes after this firmware update for me.
Hey leptun, thanks but I couldn't find anything useful there as my problem is different.

m-cas, 2020-03-08:
With the latest production fw both on the card and on the printer I am fairly happy. It is true that the transfer speed could be better but I can live with it...

closed 2023-09-26 by github-actions[bot]