prusa3d/Prusa-Firmware #1686: LCD stays blank

tifrei, 2019-04-02:
Hallo, 
i got a einsy rambo form ultimachine and put prusa firmware for my MK3 on it. 
And i had to replace the screen as i broke mine.... 

Now the LCD stay blank whatever i do. I tried flashing the new Firmware. 
When connecting via Pronterface i get the following: (anybody now whats wrong here?)

Connecting...
start
Printer is now online.
echo: 3.7.0-2201
echo: Last Updated: Apr  1 2019 20:56:14 | Author: (none, default config)
Compiled: Apr  1 2019
echo: Free Memory: 2037  PlannerBufferBytes: 1392
echo:Hardcoded Default Settings Loaded
adc_init
CrashDetect ENABLED!
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
FSensor ENABLED
Command not found!
echo:SD init fail
Wizard state: 0echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user

leptun, 2019-04-06:
Things I would check:
-I would press and hold the ok button while the firmware starts to enter the factory reset menu. It should beep loudly when it enters the mode. After that release the button and select all data.
- check if the wires were plugged incorrectly. Both at lcd assembly and printer electronics.

Could you please post a video of the issue?

NielschB, 2019-04-06:
Check the LCD contrast setting, small potmeter on the back of the LCD PCB. Happened to me with China LCD ;)

closed 2023-09-26 by github-actions[bot]