prusa3d/Prusa-Firmware #1687: [3.7.0-RC1] bed preheat issue (again)

Dasherrr, 2019-04-02:
Bed preheat error when using ABS + fan on; you can go around this error when you heat up the bed to 110 °C right from the start. Anyways, it was said this bug is fixed in 3.5 - but it's not. It's still very present in the 3.7.0-RC1. Original i3 MK3.

ThaliaFromPrussia, 2019-04-04:
Second that. Ran into the same issue. Can be prevented if I change the temp for ABS from second layer on to 100deg as well.

michalxfanta, 2019-04-04:
Bed preheat error fix for MK3 https://github.com/prusa3d/Prusa-Firmware/pull/1693

skipzero, 2019-04-05:
I'm experiencing the same on my Mk2.5 in 3.6 and the 3.7 release. And the annoying 'update your firmware' message makes it impossible to use Octoprint...

hotzy19, 2019-04-08:
I can confirm i also had this issue. mk2.5 firmware 3.6 and 3.7 release. I had to downgrade to print a particular file failing at the same place in the print every-time, other files printed without problem. when downgraded the issue disappeared

hotzy19, 2019-04-08:
I've been playing around with this issue for the past couple of hours, I don't know if this will work for any body else, but when I import my stl file into slic3r and leave it in the default middle of built platform I get the bed preheat issue, If I move the location of the stl after importing it then slice it I can print it successfully.  would love to know if this works for anyone else.

fbujold, 2019-04-15:
History 										12 april 2019

MKS2.5 firmware 3.6
slicer 1.42 beta1
Printing MK3 fan shroud. failed 4 times yet. Lays down 2 layers then fail "BED PREHEAT ERROR"
did the thermistor check as per recommended 
[](https://help.prusa3d.com/article/tn73HdQbUL-preheat-error-heatbed)
all ok, behaving as intended
Downgraded to 3.5.1-1778 as per recommendation of tech support reprinted the same fan shroud in abs. same failing Bed preheat error "while printing"

Tryied in PETG for testing.....Passed. Parts finished printing

Printed the same part in ABS with pieces of insulation on bed ....Passed. Parts finished printing

Moved around and pulled the heatbed cable, the value for the temperature stable
Secured the heatbed connections Just in case
Did a temperature and PID calibration twice PID setpoint at 223 (my favorite filament)
 
 New day  15 april

 Ambient temp 19 deg
 start of print material ABS, 255/100 initial temp then 110 for higher layers
 
Bed temp		 PSU voltage      bed voltage	            PSU DC Current
89				12.07		   10.8			   11.2
92				12.07		   10.8			   11.1
:
:
98				12.07		11.1			4.0 (PID Kick in)
99				12.07		11.2			5.2
100				12.07		11.1			4.0
105				12.07		11.1			5.2
Failed bed peheat error

Repeated the print with same scenario values similar

Temp at failure			elapsed time after printer command to raise temp
104						1:09
106						2:20
105						1:44
103						1:10

My suspicion of failing PSU was wrong
PSU is rated 20 amps drawing only 12 at the most, voltage stable
The other theory of lack hack cable extension resistance causing a voltage drop is also wrong, since voltage stable at 12 vdc

15:34
Changed bed thermistor as per tech support
Did temperature calibration
Failed again, same failure.

aboettger-tuhh, 2019-04-28:
I have similar problems. After upgrading to MK2.5 I made a printout with PET without any problem.

Now I try a print with PLA and get the PREHEAT ERROR. Both with firmware 3.7.0 and with 3.7.1-RC1. If I adjust the fan from 255 to 100 after starting the fan via the tuning menu, then the problem for the current printout no longer appears.

Terminator4, 2019-05-06:
I am faced with the same - reproducable problem: BED preheat issue
Printer: MK2 MMU2 updated to MK2.5s MMU2s
Since Firmware 3.5 i got this issue and did some research. This problem was adressed several times and i noticed an update on Firmware V3.5.2 (1.0.4) MK3/MK2.5 MULTI MATERIAL 2.0 from February 20, 2019.
> Stated improvements:
> - Heatbed regulation improved - reduced the PSU clicking noise
> - Improved heater temperature regulation
> - Mintemp updated (MK2.5 only)

**But** i also found this comment on Github by inquisitiveKangaroo on 21 Dec 2018:
So AFTER the Firmware update V3.5.2!
> Post von inquisitiveKangaroo 21.Dez.2018:
Antient Bug, with hot fix. Should be fixed in FW 3.5.1.
Later post:
>inquisitiveKangaroo comment on 21.Dec.2018:
Hi, thanks for pointing out... You are right (my bad, sorry)... we are going to fix it in next release. (inside communication error.) ) However the solution is in progress and we will try to make it up and running asap.
Later post:
>inquisitiveKangaroo commented on 9.Jan19:
Hi, This issue is not as trivial as it may appear. Currently we have a beta version solving this issue. We are going to run some basic tests during this week. If everything goes well, it will probably appear on github in a week or so. Sincererly, Kangaroo
Later post:
>inquisitiveKangaroo commented on 1 Feb
Hi, the solution has been already merged. #1460 You can compile it by your own or the new release should be out in a week. Have a nice printing day.) Kangaroo

Up to now, i missed an update notice on the official firmware page, that these improvements are introduced. Can you confirm, that the improvements for MK2.5 users with MMU2 are really integrated ?
Or could you provide a target date for the firmware update ?
Thanks in advance!

DevTown, 2019-05-12:
Got that error Today. Printed several items with PLA everything OK. Now with a PETG test after 2nd or 3rd Layer i get "bed preheat error" on the LCD . Sliced with Slic3r-PE PETG settings that came with it. 
FW 3.7.0 on Prusa 2.5

closed 2023-09-26 by github-actions[bot]