prusa3d/Prusa-Firmware #1688: Please restore previous load/unload temperature behavior

Commod0re, 2019-04-02:
The new filament load/unload preheat requirement is obnoxious and wasteful, why must I wait for a preheat when the nozzle is already 195C? Why does that preheat trigger a bed preheat? Please restore the old behavior of simply allowing the extruder to move as long as the nozzle is still above a certain temperature.

m-cas, 2019-04-03:
Agreed. Personally I'd prefer the 2nd choice: leave the thing as is but do not heat up the bed.

Panayiotis-git, 2019-04-03:
I also dissagree with the heat-bed preheat. When I want to remove the filament I do not want to wait for the heat-bed to also warm up. For this I use the preheat menu.

leptun, 2019-04-03:
You don’t have to wait for the bed. Once the hotend reaches the target temp - ~10deg you can press to remove filament.
I agree with you that the bed does not need to be heated though. I will try to implement this change.

ctilley83, 2019-04-03:
It should not heat the bed.  All I want to do is load filament sometimes.

michalxfanta, 2019-04-04:
We think about it.

Crowlord, 2019-04-04:
My tuppence worth

Loading heat the bed
Unloading dont.

timur-tabi, 2019-05-18:
The original behavior was better.  Heating the bed is pointless, and we shouldn't need separate temperatures for different filaments.  Just heat the nozzle to around 200 degrees and then unload or load.

bpvarsity, 2019-05-18:
I like the bed heating up. They just need to make it a user selectable option for those that don't want it

timur-tabi, 2019-05-18:
Bed temperature has nothing to do with load/unload, and there are already too many options.  It used to be so simple - click unload, and then it heats the nozzle and unloads.  There was nothing wrong with that behavior.

Panayiotis-git, 2019-05-18:
If I want to use the preheat the bed I use the Preheat option. The heating of the bed during loading/unloading is pointless.

m-cas, 2019-05-18:
I share exactly the same viewpoint. 

-
From my phone while mobile, please excuse any errors.

ttabi, 2023-09-18:
This issue is not stale.

3d-gussner, 2023-09-19:
@ttabi This has been solved with https://github.com/prusa3d/Prusa-Firmware/pull/2335.

closed 2023-09-19 by 3d-gussner