prusa3d/Prusa-Firmware #1694: [MK2.5 3.7.0]Crazy screen when compile FW from source

dronecz, 2019-04-04:
I downloaded FW source code and compile it without any modification and screen is full of [crazy letters.](https://youtu.be/ylnCZqq7D8Y). Same behavior was with RC1.

When I flash HEX file for MK2.5, everything is OK. When I compile older FW, printer is also working normally.

leptun, 2019-04-04:
Did you set “LANG_MODE” to 0 in config.h?

It is required for mk2.5 firmware to display corectly

dronecz, 2019-04-04:
Yes, you are right. I forget about this 😞 . 

Solved, but would be better if there is check what board you are using.

closed 2019-04-04 by dronecz

leptun, 2019-04-04:
Glad to be helpful. If you use the PF-Build.sh script on linux or windows with git bash you have it toggled based on what firmware you are compiling. You can see how to use it in the latest readme lower.

dronecz, 2019-04-05:
>  If you use the PF-Build.sh
When I edit something in FW, I just compile it and upload it straight away to printer so make no sense for me to use build script 😉 . I will try to make change to FW and try pull request.