prusa3d/Prusa-Firmware #1695: Bug/Feature? - Resume after pause and power off like in power outage

quarkcz, 2019-04-04:
When power goes off, printer will resume printing.

When I manually pause print and then power goes off, it DOES NOT resume. 

I wanted to move printer to another room. I know about power outage function so I didn't worry about my print. I paused printing (to be more gentle to my printer), powered off, moved printer, plugged in, but no resume poped up. Whole print was lost.

So now I know, that if I want to move printer, i can't pause print, I have to switch off power during print and them move it. It feels very illogical to do it this way, but seems to be only way possible today. Could you please fix this?

mkbel, 2019-04-09:
Reproduced.

unixunion, 2020-09-02:
Is there progress on this? When I have MMU failure, it sometimes gets stuck on resuming position..., and doing a power cutout and back in again doesn’t resume

wavexx, 2020-09-04:
It's a know limitation of the PP function currently #2325, #2249, #2001

DoctorMarshall, 2021-06-01:
Reproduced.
Can we have a ETA of this? Seems like it could be a quick fix and is hanging for 2 years already.

Thanks!

closed 2023-07-18 by DRracer