prusa3d/Prusa-Firmware #1698: MMU2S Feature Request - Enable/Disable From LCD Menu

Eddiiie, 2019-04-05:
I could not find this previously posted.

It would be awesome to have a menu item on the LCD screen to enable/disable the MMU2s.  Sometimes I just want to print something quick.  Or, leave all the filaments loaded in the mmu, remove the PTFE tube to the hotend and stick some filament in there and go.

Currently, I have to unplug the power and signal cables to disable the MMU2s and that is just a pain.

Maybe this is more of a main firmware request rather than a feature request in the MMU2S firmware.

Thank you.

michalxfanta, 2019-04-05:
Hello, thanks for the idea, we'll consider implementation.

leptun, 2019-04-07:
I implemented it, but I have no way of testing if it works.
If you want you can test it yourself.

For now you can toggle this in the Settings menu when mmu is connected or it is disabled from menu.
VERY IMPORTANT! : You have to reset the printer after you change the setting. Otherwise nothing happens. I intend to add an auto reset using the watchdog timer later.

Attached you can find hex files with compiled firmware. If you want to compile it yourself, you can clone this: https://github.com/leptun/Prusa-Firmware/tree/MK3_disable_MMU_from_lcd

[FW370-Build2201-desable_MMU_from_lcd.zip](https://github.com/prusa3d/Prusa-Firmware/files/3051381/FW370-Build2201-desable_MMU_from_lcd.zip)

3d-gussner, 2019-04-07:
Disabling the MMU2s might seam to be a nice idea, but if the MMU2s is disabled it will not communicate with the RAMBo board
  - MMU2s is dumb and all the brain is on the RAMBo
  - Selector if it is not powered it can move and filament might get stuck or produce lot of drag
Also you the IR sensor triggers differently with the MMU2s Extruder body ... so as a standalone MK3s it will expect that the IR sensor triggers few mm before the bondtech gears for autoload. This may generate few issues.
A bigger issue you will have if the filament runs out on a standalone MK3s with MMU2s Extruder body, as it will trigger when the filament already (maybe) nearly passed the bondtech gears, which makes it impossible (maybe hard) to unload the last part and you have to try to get the filament manually out.

There are other things can going wrong with that.

On the other side if you unplug the MMU2s tube from the Extruder and feed the filament as on a standalone MK3S WITH filament runout sensor disabled, it would work fine.

Eddiiie, 2019-04-10:
@michalxfanta  I know you can use Single mode but the point of disable is to just unplug the orange feed tube, leaving your other filaments loaded,  load another filament on another spool holder, send print and go (to work) ... It can be 15 minutes or so to heat/unload, rewind, rethread, heat, load/feed new filament, pre-heat bed, print.   I could be at work by then!  :)

@leptun Thank you for the build.  What are your thoughts on 3D-gussner comments above?

@3d-gussner  Maybe I am confused.  Don't all MK3's  now ship as MK3S 'es?    And, isn't the 'S' revision a new filament sensor on the extruder (and firmware)?  So, if the printer boots up without detecting the MMU2S, it becomes a MK3(S) (so-to-speak), therefore, wouldn't the issue you mention with the filament not feeding enough not exist?

Interesting about disabling the filament sensor, pulling the tube, loading a filament seems to be what I am looking for..   What is stopping the MMU2 from feeding one of the other filaments it has loaded?

maxcom99, 2019-09-18:
I am looking to disable the MMU2 on my MK3S so I can leave the 5 colors fed but feed a "one off" print any time I want personally. Basically I want to bypass the hassle but also the extra tubing friction and "perfect mmu" setup requirement to print something I know works perfectly without MMU2.

rv10guy, 2019-10-17:
I agree that this is needed urgently. My MK3S was my solid “go-to” “never-fail” printer. The MMU2S has impacted that reliability. I ended up doing several mods and worked through several issue to get it into a “workable” state, but I don’t believe it will ever be as reliable as it was before. So while I did this i “lost” my printer. Had I been able to revert back to non MMU2S mode it would have really made the whole adoption process easier. Also even though I have MMU2S I still print well over 50% of my prints with a single filament. Unloading, Loading, sealing with the buffer and all of the unnecessary steps make my workflow far less efficient. 

Additionally when I print with abrasive filaments, I prefer to use my Olsen Ruby. I would honestly prefer to use it all the time, but with the MMU2S it does not form a good tip so retraction through the selector is problematic.

gitsh01, 2019-11-01:
Please add this feature. When I'm printing 8 small 50g makerbox samples, I don't want to have to feed them all the way through the buffer/PTFE tubes/extruder. I just want to do a quick old-MK3s style print. 

It's easy for me to disconnect the PTFE tube from the hot end, but very hard to reach all the way back to pull the power plugs from the MMU2 due to the lack of back access in the enclosure. Only have one front door.

JordanGrantF15E, 2019-11-23:
I'll add my voice. I'd like to be able to disable the MMU2S easily when I go back to printing single-filament prints (most of the time for me).

noppej, 2019-11-23:
+1 ... please :)

Photogad, 2020-03-26:
> I implemented it, but I have no way of testing if it works.
> If you want you can test it yourself.
>
> For now you can toggle this in the Settings menu when mmu is connected or it is disabled from menu. [...]

Did anyone test this

gudnimg, 2023-01-30:
Closing this in favor of https://github.com/prusa3d/Prusa-Firmware/issues/1876 this has been implemented in 3.13. 👍

closed 2023-01-30 by gudnimg