prusa3d/Prusa-Firmware #1700: MMU2 fails to load filament und FW 3.7.0

Gountled, 2019-04-05:
I have the strange behavior, that it does all the heating, mesh bed leveling and after that it tries to load the filament. All is fine and filament is loaded into the nozzle, but afterwards it is unloaded completely again. After a couple of time doing this, the printer tries to print from back to front instead the "normal" direction, but without any filament and the heater is also switched of.
I switched the IR sensor in the menu on and off, no change.
I switched the MMU to normal and to stealth mode, no change.
I can see, that the IR sensor reacts when the filament reaches the nozzle, but still it is retracted immediately after.

Switching back to 3.6.0 solves this problem immediately!

leptun, 2019-04-05:
Duplicate of #1689

jindrichbenes, 2019-04-08:
Have you upgraded fw of MMU unit itself?

PavelSindler, 2019-04-08:
Are you printing from SD card or is it USB print (Octoprint, Pronterface...)?

Gountled, 2019-04-08:
Usually I am printing via Octoprint.
Both have been Upgraded, printer firmware and also MMU2S

fnordsh, 2019-04-14:
I had the same issue. I thought my IR sensor was calibrated well, but shifting it a tiny bit to the right solved the issue for me.

Gountled, 2019-04-15:
I will give it a try, but firmware 3.6 works like a charm.

fnordsh, 2019-04-15:
I think 3.7 uses the IR sensor more extensively, which is probably a good thing (once it's calibrated right).

Gountled, 2019-04-16:
So I flashed 3.7.0 today aigain and moved the IR sensor as much to the right as it was possible.
The result is: It loaded the filament as it should! Thanks for the advice!

Printing single colour calibration cube right now (from SD Card). Next test will be a dual colour print.

fnordsh, 2019-04-16:
Nice! Glad I could help!

closed 2019-04-18 by michalxfanta