prusa3d/Prusa-Firmware #1702: Timeout for New Firmware Available message

Barnabasek, 2019-04-06:
I would like to request a timeout for the "New Firmware Available" message that appears on the LCD screen when you use newer Slic3r/Firmware than what is on the printer. The message currently requires user intervention (physical pressing of a button) which completely negates usage of any remote control like Octoprint. Not to mention it breaks the "flow". I can't count how many hours I wasted after starting the print from SD card "as usual" and then coming back later only to see that unconfirmed message on the screen instead of a finished print.

Can this "New Firmware Available" message simply time-out after let's say 5-10 seconds instead of physically pressing the button?

stahlfabrik, 2019-04-06:
Then why not deleting that unnecessary line from the start gcode in your slicer?

Barnabasek, 2019-04-06:
> Then why not deleting that unnecessary line from the start gcode in your slicer?

Yes, that would be a workaround. But not a real fix that I am proposing here.

stahlfabrik, 2019-04-06:
Whatever works for you:-)

alexwhittemore, 2019-04-06:
I'd add that the whole notion of a firmware update being important enough that the default expected action is "carry out the update" comes from software and in particularly security, where this case would seem to be wholly the opposite. While an updated firmware might deliver nice new features, it's almost never critically necessary, and as frequently as it might address an important safety concern, it might also introduce one (https://github.com/prusa3d/Prusa-Firmware/issues/1636). Further, by virtue of changing printer behavior sometimes substantially (many users appear to report that the new 7x7 leveling necessitates a z height recalibration), it's usually safe to assume a firmware update will require substantial validation effort from the user. For these reasons, I don't think the assumption should hold that firmware updates represent an expected immediate user action, and deliberately breaking a user workflow is very much not the right course of action.

TL;DR firmware updates aren't nearly important enough to have a must-confirm user dialog.

closed 2023-09-26 by github-actions[bot]