prusa3d/Prusa-Firmware #1709: [3.7.0] MMU2s Filament load failure causes idler barrel to crash - No way to save print

TimothyGold, 2019-04-08:
I started and almost finished a MMU2s Print last night but right towards the end it attempted to load a filament but for some reason it ran to the nozzle and crashed and when I found it stalled the LCD stated at the bottom "Failed MMU2 Load" or something like this.  The Nozzle was cold and the filament was still loaded to the bondtech gears.  I don't know exactly what caused this but normally when there is a failed MMU2 load it gave the options to reheat fix and continue.  This time no menu to continue.  I had to open the MMU2 and found that the MMU2 pushed and coiled up the filament at the top so it was pretty stuck.  When I removed the filament I hoped I could save the print but found the Idler barrel was off and I don't know of a way to rehome this in the middle of a print.  I also did not have the option to reheat the nozzle.  I ended up resetting the MMU2 to force a rehome but then the MK3s started to print in mid air without the nozzle being heated or any filament loaded.   Anyway long story short the print failed.....  The only thing I can conclude to is that the nozzle somehow went cold and this may have caused the crash when the filament tried to load to the nozzle.  I would think there should have been a check for this and not try to load the filament if there was any kind of a heat issue.  After pulling the print off I tested the nozzle and it heated fine and does not seem to have any problem.  I guess I will have to spend more time to make sure but there might be a check that is missing in this process.

mkbel, 2019-04-08:
> When I removed the filament I hoped I could save the print but found the Idler barrel was off and I don't know of a way to rehome this in the middle of a print.

Just push the reset button on MMU to rehome everything.

TimothyGold, 2019-04-08:
@mkbel I did push the reset button on the MMU as a last resort but like I stated that once I did this then gcode continued to process from Octoprint and printed in air without loading the filament that was in error.  There is definitely an issue with trying to reset the MMU while in this state.  I did state above that when this filament failed to load I did not get an option to reheat the nozzle like I have in all other versions of the firmware before.  This time it did move off to the Left like it has before and the nozzle heat was shut off but with a status message at the bottom of the LCD stating that the filament failed to load.  No option to continue from the MK3s.

mkbel, 2019-04-08:
Ahh, Octoprint. This is known bug.

mkbel, 2019-04-08:
I am working on the fix at the moment.

TimothyGold, 2019-04-08:
OK.  Thanks for the info.

mkbel, 2019-04-08:
It is already partially fixed there. https://github.com/mkbel/Prusa-Firmware/tree/fix_octoprint_mmu_load_failed but the fix is not complete, in case it fails again after you resume print you get missing layer.

TimothyGold, 2019-04-08:
That's awesome.  I know Octoprint presents a brand new lot of challenges but Octoprint is a must in my setup.  I cannot do it another way right now.  Thanks again for working on this...

mkbel, 2019-04-09:
Fixed in #1713

closed 2019-04-09 by mkbel