prusa3d/Prusa-Firmware #1714: Please Update MK3S Firmware Build Instructions

YoungR0se, 2019-04-10:
MK3S firmware is compiled in Arduino 1.8.5 (https://github.com/prusa3d/Prusa-Firmware)

Arduino 1.8.5 _no longer_ has a line of text in platform.txt that can be changed to read 
"compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--gc-sections", because that line now reads in 1.8.5 as
"compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections";
therefore editing and compiling firmware for the MK3S using Arduino 1.8.5 does not work.

And before anyone suggests it, I have read the readme file, and it is outdated as well.  Can someone please tell me what steps to take so that the MK3S firmware compiles correctly?

Fair farren.

mkbel, 2019-04-14:
I know that building is very unpleasant for Windows users now, but we have no capacity at the moment to make it more pleasant.
Did you try to follow Readme.md and failed? What kind of error message you get?
If you followed step you mentioned.
If your line reads
`compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections`
after add "-Wl,-u,vfprintf -lprintf_flt -lm" to "compiler.c.elf.flags=" before existing flag "-Wl,--gc-sections"
it should look 
`compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--gc-sections`

YoungR0se, 2019-04-14:
Hi mkbel,
In E:\arduino-1.8.5-windows\arduino-1.8.5\hardware\arduino\avr, platform.txt contains the exact line of code you wrote above (I had tried that already, but copy/pasted your code to try) and the firmware compiles without errors, but on the printer itself comes garbled as you can see in the attached photo.
[image]

If I return platform.txt to its original state (compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections) The home screen of the printer is normal, but Z height is '?' instead of '0.00'.  
If I use pre-compiled .hex file flashed with Slic3r, no problem, Z height is 0.00.  But i can't use this firmware, because the Z axis of my printer is 350mm.  

What environment does Prusa use to build the firmware in the first place?

3d-gussner, 2019-04-14:
Try the PF-build.sh script, which downloads a known build environment and guides you through the compiling process.
Read from https://github.com/prusa3d/Prusa-Firmware#using-linux-subsystem-under-windows-10-64-bit
To prepare your Windows computer to use PF-build.sh . If you prefer using git-bash follow these instructions https://github.com/prusa3d/Prusa-Firmware#using-git-bash-under-windows-10-64-bit

mkbel, 2019-04-15:
I would suspect omitting step 2.c.: In file "Firmware/config.h" set LANG_MODE to 0.
If you didn't omit that step, try to flash stock firmware, switch printer to English language and then try again flash what you built and let me know, if that helped.

> What environment does Prusa use to build the firmware in the first place?
Our production builds are 99.9% equivalent to https://github.com/prusa3d/Prusa-Firmware#linux this is also easiest way to build as only one step is needed - run single script, which downloads patched Arduino from github, builds using it, then extracts translated strings and creates language variants (for MK2x) or language hex file for external SPI flash (MK3x). But you need Linux or Linux in virtual machine. This is also what happens when you open pull request to our repository - all variants are built by Travis http://travis-ci.org/ (to check for compilation errors). You can see, what is happening in .travis.yml. It would be also possible to get hex built by travis, only deploy step is missing in .travis.yml. You can get inspiration how to deploy hex by travis and how to setup travis in https://github.com/prusa3d/MM-control-01/ repository. Final hex is located in ./lang/firmware.hex

Community reproduced this for Windows in https://github.com/prusa3d/Prusa-Firmware#using-linux-subsystem-under-windows-10-64-bit or https://github.com/prusa3d/Prusa-Firmware#using-git-bash-under-windows-10-64-bit .

YoungR0se, 2019-04-15:
mkbel, if I leave the LANG_MODE unchanged I get the scrambled home screen again.  As I said in the above post, flashing stock firmware works perfectly, flashing edited firmware in Arduino 1.8.5 give Z height '?' I will try the Linux method.  
Thanks for all your input :)

mkbel, 2019-04-15:
If the Z height is '?' it means step:
after add "-Wl,-u,vfprintf -lprintf_flt -lm" to "compiler.c.elf.flags=" before existing flag "-Wl,--gc-sections"
wasn't successful.
I added these two common problems into FAQ section in readme.md. Probably you have updated Arduino and now platform.txt used resides somewhere in your [home](https://en.wikipedia.org/wiki/Home_directory#Default_home_directory_per_operating_system) directory probably folder .arduino15

3d-gussner, 2019-04-15:
I think using pf-build.sh under linux, Windows git-bash or Windows with linux subsystem would prevent any arduino ide issues, updates, libraries, as it downloads a known build environment.

YoungR0se, 2019-04-15:
mkbel, I have searched both hard drives for platform.txt, then searched each text file found both electronically and manually for "compiler.c.elf.flags=".  Only one file has this particular text, so that is the one I edited as you suggested to read "compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--gc-sections".  This doesn't alleviate the '?'

YoungR0se, 2019-04-15:
I have also, in Tools → Boards Manager → Arduino AVR Boards by Arduino, tried version 1.6.11 (this was for when Arduino 1.6.9 was used, and the only place I can find the Board version mentioned) and version 1.6.21, 1.6.22 and 1.6.23, all of which produce:
[image]
Tomorrow I will install Linux subsystem.

YoungR0se, 2019-04-17:
I'm now in the process of loading a Linux subsystem.  For any other poor sucker out there reading this, this page is very helpful : 
https://www.lifewire.com/ultimate-windows-7-ubuntu-linux-dual-boot-guide-2200653

3d-gussner, 2019-04-17:
> I'm now in the process of loading a Linux subsystem. For any other poor sucker out there reading this, this page is very helpful :
> https://www.lifewire.com/ultimate-windows-7-ubuntu-linux-dual-boot-guide-2200653

You **don't need a dual boot system** to compile the firmware on a Windows 10 64 bit computer.
Windows Linux subsystem IS NOT a dual boot Linux and Windows... it is more like a intergrated Linux VM for Windows.

Please read the README it has step by step instructions to prepare your Windows 10 computer.
**There is really no need to shrink your partition and so on and get a dual boot system.**

- https://github.com/prusa3d/Prusa-Firmware#using-linux-subsystem-under-windows-10-64-bit for subsystem linux
- https://github.com/prusa3d/Prusa-Firmware#using-git-bash-under-windows-10-64-bit for "git-bash" users

YoungR0se, 2019-04-17:
Hm, good for you. **I have Windows 7**. I did a few hours research yesterday concerning Linux and Windows 10/Windows 7 and now have a partitioned SSD (without shrinking, just moved the pagefile) so I can boot to Linux or Windows 7 on the same laptop without any hassle.  
The irony is that Windows 7 will be phased out as of next year, so getting uesd to Linux is probably a good thing :)

3d-gussner, 2019-04-18:
@YoungR0se Sorry didn't know you are using Windows 7. Did you try to update to Windows 10? I could do a free update on an 'old' computer few weeks ago, even officially it is stopped.
Otherwise: Good luck with the dual boot system.

YoungR0se, 2019-04-18:
No problem, I have no intention of 'upgrading' to windows 10.  Thank you for all your support in the mean time, I will post my eventual solution here, when I have found it.  Things are looking better, but it's going to take a little time to get everything running smoothly.

YoungR0se, 2019-04-20:
Ok, I got things working and successfully upgraded the printer firmware to the latest, including my 350mm modifications.  This is what happened:
Replace Windows 7 with Linux Mint.  Download Prusa firmware from GitHub and follow the (2 paragraphs of) instructions regarding Linux builds.  Perform any modifications to the firmware in each individual file (not in Arduino) - edit the height in Variant/MK3S and in ultracpp.  Run PF-build.sh as executable file and follow all prompts.  Flash resultant .hex using executable SlicerPE.AppImage.  Flash Prusa-provided MMU2 .hex like normal.

YoungR0se, 2019-04-20:
Seems so simple, doesn't it.  Took two days to make everything work, but it's working now.

YoungR0se, 2019-04-20:
This is also very important: https://github.com/prusa3d/MM-control-01/blob/master/README.md#linux-1
And so is this: https://www.downtowndougbrown.com/2016/10/fix-for-usb-serial-port-being-opened-by-modemmanager-at-startup/

closed 2019-05-12 by YoungR0se