prusa3d/Prusa-Firmware #1715: MMU2S Feature Request - Please Add Beep When MMU2S Needs Attention

Eddiiie, 2019-04-10:
Hi,
While I work out my filament feed issues and placement of the buffer, tubes etc, I have been getting a lot of MMU Needs Attention messages.  However, it is a long time before I notice it since I am in the other room.   It would be good it the printer would beep for attention at least once.

Thank you for your consideration.

thiloroehl, 2019-05-10:
I would love to have this inside octoprint using pushover. I already see something like *requires user intervention* in the terminal window coming from the printer. I'm sure it would be possible to already use this to trigger events on pushover.

JohnnyDeer, 2021-01-19:
Hi @Eddiiie, MMU notification sounds are already implemented, I'm closing this issue.

closed 2021-01-19 by JohnnyDeer