prusa3d/Prusa-Firmware #1717: Feature Request - Add a fade factor for mesh leveling

rafaljot, 2019-04-10:
I'm trying to add ENABLE_LEVELING_FADE_HEIGHT feature, like in Marlin firmware.
https://github.com/MarlinFirmware/Marlin/pull/5299

I added to planner.cpp :

#define MESH_CORRECTION_FADE_HEIGHT 3
#ifdef MESH_CORRECTION
  #define MESH_CORRECTION_EXPR(X,Y,Z) (Z > MESH_CORRECTION_FADE_HEIGHT ? 0 : mbl.get_z(X, Y)*(MESH_CORRECTION_FADE_HEIGHT-Z)/MESH_CORRECTION_FADE_HEIGHT)
#else
  #define MESH_CORRECTION_EXPR(X,Y,Z) (mbl.get_z(X, Y))
#endif

then use  +MESH_CORRECTION_EXPR(x,y,z) instead +mbl.get_z(x, y)

planner.cpp , line 749:
#ifdef MESH_BED_LEVELING
      if (mbl.active ){
          target[Z_AXIS] = lround( (z + MESH_CORRECTION_EXPR(x,y,z)) * cs.axis_steps_per_unit[Z_AXIS]);
      }else{
          target[Z_AXIS] = lround(z*cs.axis_steps_per_unit[Z_AXIS]);
      }
#else

I think It works. 
Does it make sense?

michalxfanta, 2019-10-11:
We are planning to add this feature to one of the next releases.

rdpeake, 2021-01-21:
was this feature ever added?

closed 2023-09-26 by github-actions[bot]