prusa3d/Prusa-Firmware #1718: 3.7.0.Final - Failing Calibration

sopis, 2019-04-10:
Hello

I just updated my perfectly calibrated Prusa MK2.5S BEAR with 0.04 skew to FW 3.7.0.
1. First of all, the 7x7 mesh bed leveling did not trigger - it was standard 9 points calibration
2. Cannot finish XYZ calibration - after all 4 points are probed, it will state XYZ calibration failed (even though the machine was perfectly calibrated before FW upgrade)

michalxfanta, 2019-04-11:
Hello, 
1. You have to change 3x3 to 7x7 at _Settings > Mesh bed leveling_

sopis, 2019-04-11:
Hi Michal

Thanks for the info.
What about the calibration?

michalxfanta, 2019-04-11:
Could you send to us logs from XYZ calibration? Connect printer with USB cable, run Pronterface, connect printer on baud rate 115200, run XYZ calibration again. Thanks.

sopis, 2019-04-11:
Ahoj Michal

Diky za tip...hned ako pridem domov to pozrem a poslem log..
Co je mi zvlastne, ze predtym nez som tlaciaren upgradol, kalibracia bola
uplne perfektna, na 0.04 skew...

Uvidime teda...

Dakujem!

sopis, 2019-04-11:
Ahoj Michal

Posielam 2 logy.
1. Znova som dal upgradovat FW
2. XYZ kalibracia

avrdude: Version 6.3-20160220-prusa3d, compiled on Feb 11 2019 at 18:25:11
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:/PROGRA~1/Prusa3D/Slic3rPE/resources\avrdude\avrdude.conf"

         Using Port                    : COM7
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: hfuse reads as D0
avrdude: safemode: efuse reads as FD
avrdude: reading input file "D:\Documents\3D PRINTING\!MODS\FW\3.7.0\prusa3d_fw_MK25S_3_7_0_RAMBo13a_en-cz.hex"
avrdude: writing flash (227726 bytes):
avrdude: stk500v2_command(): command failed

Writing | ################################################## | 100% 41.39s

avrdude: 227726 bytes of flash written
avrdude: verifying flash memory against D:\Documents\3D PRINTING\!MODS\FW\3.7.0\prusa3d_fw_MK25S_3_7_0_RAMBo13a_en-cz.hex:
avrdude: load data flash data from input file D:\Documents\3D PRINTING\!MODS\FW\3.7.0\prusa3d_fw_MK25S_3_7_0_RAMBo13a_en-cz.hex:
avrdude: input file D:\Documents\3D PRINTING\!MODS\FW\3.7.0\prusa3d_fw_MK25S_3_7_0_RAMBo13a_en-cz.hex contains 227726 bytes
avrdude: reading on-chip flash data:
[... 648 lines cut ...]
OK
 x0=751
 y0=19647
xyzcal_find_point_center2
 x0=751
 y0=19647
 z0=-534
   z0=-535
OK
 x0=750
 y0=19645
xyzcal_find_point_center2
 x0=750
 y0=19645
 z0=-535
   z0=-537
OK
 x0=750
 y0=19645
xyzcal_meassure_leave
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: paused for user
echo:busy: paused for user

sopis, 2019-04-11:
To compare, I flashed back to 3.6.0 and did the calibration successfully - log attached.
Results:

Y distance from min:
Left: 8.92mm
Right: 8.63mm

Measured skew: 0.07
Point offset:
X -4.92mm
Y 0.43mm

[XYZ_Calibration.txt](https://github.com/prusa3d/Prusa-Firmware/files/3069871/XYZ_Calibration.txt)

cimoalpacino, 2019-04-12:
Your calibration points are way off.
it shoud be {x,y}:
{12, 6} ; {220, 6} ; {220, 198} ; {12, 198} mm
snippet from xyzcal.cpp: 
const int16_t xyzcal_point_xcoords[4] PROGMEM = {1200, 22000, 22000, 1200};
const int16_t xyzcal_point_ycoords[4] PROGMEM = {600, 600, 19800, 19800};

Your point are:
{7.29, 4.53} ; {215.14, 4.21} ; {219.39, 203.78} ; {7.50, 196.45} mm

If you want your calibration to pass, you need to edit xyzcal.cpp to your values
const int16_t xyzcal_point_xcoords[4] PROGMEM = {729, 21514, 21939, 750};
const int16_t xyzcal_point_ycoords[4] PROGMEM = {453, 421, 20378, 19645};

Cheers.

Ipirk88, 2019-04-16:
I have the same problem:
Failing the xyz or z calibration - but only when i set the Steps/mm for the Z axis not to 400.

When i set M92 Z399.48 then it fails,
when i set M92 Z400 then the calibration is  successfuly.

Is the same on MK3S and MK2.5S
FW bug?

sopis, 2019-04-23:
@cimoalpacino 

Hi - thanks a lot for your research and insight.
Unfortunately, even though I understand your reply, it is rather too advanced for me...
I don't understand why would my calibrations be off - on previous FW (3.6.0) , I had no problem calibrating...

Could you please explain what it means they are off? Could this be because I have a BEAR upgrade?

As for editing the xyzcal file - the printer actually prints quite well...I see some ghosting due to vibrations (also need to figure that out), however it really prints rather fine...so why am I seeing these issues and should I actually be concerned?

Thanks!!

cimoalpacino, 2019-04-23:
Hi.
If you assembled everything correctly, you should be fine.
But to me, there are 2 options:
- you maybe assembled something wrong (check all distances as per bear drawings) 
- you flashed the wrong hex or config file. 

Try to reflash the firmware with the modification mentioned above and see if it passes the test. 

Xyz calibration may find all points, but if they are too far from what is set in xyzcal_point_xcoords[4] and const int16_t xyzcal_point_ycoords[4], then the calibration fails.

Otherwise your printer will print just fine and everything will work, the only think that will be disabled is skew correction. 
Skew correction will be enabled once you pass xyz calibration. 
Cheers.

gregsaun, 2019-04-25:
The Bear frame use the same position as the Original Prusa frame so that should not be a problem.

However, since MK2.5S/MK3S, the original extruder went smaller in height and Prusa updated the calibration to increase Z height. The Bear extruder for MK2.5S/MK3S currently in development is not going to follow that road for two main reasons: print quality and angled fan. Once the release of this extruder is going to be online it will be explain clearly what procedure to apply to get XYZ calibration.

Finally, lots of people have reported 3rd and 4th point calibration error since firmware 3.1.4 when Prusa introduced a new XYZ calibration algorithm, including on stock frame. Several people have identified were this issue comes from and have reported to Prusa but we still have no fix for that.

cimoalpacino, 2019-04-25:
Hi,

if Z height is the problem, you just need to edit one piece of code in `ultralcd.cpp`.
Look for the code below and edit the value after Z_MAX_POS.
If Prusa added Z height, then you decrease the number.
Example: if the Z travel distance is 2mm higher, then instead of `Z_MAX_POS+4`, you use `Z_MAX_POS+2`.  

    // Let the machine think the Z axis is a bit higher than it is, so it will not home into the bed
    // during the search for the induction points.
	if ((PRINTER_TYPE == PRINTER_MK25) || (PRINTER_TYPE == PRINTER_MK2) || (PRINTER_TYPE == PRINTER_MK2_SNMM)) {
		current_position[Z_AXIS] = Z_MAX_POS-3.f;
	}
	else {
		current_position[Z_AXIS] = Z_MAX_POS+4.f;
	}
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
return true;`

fbujold, 2019-04-25:
Can you give the therorical zmax value. I Do mot know if i Shall go plus or minus un the zmax value

cimoalpacino, 2019-04-25:
When you do xyz calibration and your nozzle is far from bed (you can clearly see it when it's circling above the bed), you need to lower the current_position[Z_AXIS] value. Because you can't change Z_MAX_POS, you just change (lower) the number after the plus sign.

brotherdust, 2019-05-09:
My printer (MK3S) is now failing as well. I don't really understand why it's doing this now. =/ All I did was move it to another location. I have attached the output of the XYZ calibration routine below. Can someone please have a look and tell me why it's failing?
Thanks!
[XYZ_calibration_log-2019-05-09.txt](https://github.com/prusa3d/Prusa-Firmware/files/3163696/XYZ_calibration_log-2019-05-09.txt)

cwbullet, 2019-09-14:
Has Anyone solved this?

closed 2023-09-26 by github-actions[bot]