prusa3d/Prusa-Firmware #1727: Print quality issue at Seam using external first setting

TraylorHead, 2019-04-12:
Since upgrading to 3.6.0 and 3.70 I have has this issues.

- Tried full factory reset & calibration.
- Issues show with PLA and PET but more visible with PET.
- Defect looks identical on almost all of them I have tried printing.
- Tried adjusting filament temp, retraction speed

Print attached model
- External perimeter first
- MMU2 mode

Gcode generated does not look correct so don't know if it's a Slicer issues or machine issue.

Thanks
Jimmy

[image]
[image]

[Scallop Nut.zip](https://github.com/prusa3d/Prusa-Firmware/files/3075011/Scallop.Nut.zip)

mkbel, 2019-04-14:
Can you please attach also gcode?

TraylorHead, 2019-04-17:
Here you go. I have included code from Slicer and PrusaControl and pictures of what I see with both.
[2101 Nut gcode and pics.zip](https://github.com/prusa3d/Prusa-Firmware/files/3090012/2101.Nut.gcode.and.pics.zip)

Thanks for looking into this.
Jimmy

jindrichbenes, 2019-04-23:
Try printing the part with MMU2 Single mode, instead of the multi material mode (if printing in one material is what you want)

TraylorHead, 2019-04-30:
Hi tried single mode with same result, not as bad.   Take a look at gcode and you will see lines that are not in place and flying outside/inside the model.  When I slice and generate gcode using Prusa Control or another slicer I do not see these lines flying around. They could be enter/exit of the head, but still does not make since as not there on other slicers gcode.

Also I noticed at different layers the model is broken apart. Look at top, just left of center.

I also recreated the model using right angles instead of a scalloped look,

[image]

TraylorHead, 2019-04-30:
Also please not it does it on older machine firmware, but does not to seem to be as bad.

jindrichbenes, 2019-04-30:
Try a different software for the gcode preview. The outdated prusa control does not preview new gcodes correctly.

TraylorHead, 2019-04-30:
I tried vewing it in Simplify3D and it looked correct, but still does not explain the issue I'm having.

Tonight I will try resetting the printer settings in Slicer and use the default settings. I will also try resetting the printer to defaults again.

I did notice that other slicers are aligning the seam, I will try that also.

I will let you know.

closed 2023-09-26 by github-actions[bot]