prusa3d/Prusa-Firmware #1730: Runout detection cannot be enabled

jsprenkle, 2019-04-14:
In firmware version 3.7 runout detection is disabled and cannot be enabled. It worked flawlessly for me in the past. Please allow user control of this option. Removing a feature that worked after purchase is NOT cool.

ochm, 2019-04-15:
Do you mean Filament Sensor? Or what is Runout? 
I'm using 3.7.0-2201 without problems (filament sensor works OK], and appreciated 7x7x5 mesh bed leveling.

jsprenkle, 2019-04-15:
Started a print with a small remainder of filament. Printer did not detect end of filament and continued to try to print with no filament. Tried to enable filament sensor from menu. LCD screen says it's disabled. Worked perfectly in the past.

ochm, 2019-04-15:
Try to do selftest, check sensor cable connection. Disabled sensor (N/A) means that the sensor not found, IMHO. Or try to do downgrade?

Prusa-Support, 2022-05-24:
Hello! Sorry for intervening on this thread so late. Did you refer to the filament sensor on MK3, non-S? Did it then work after a firmware downgrade, or with more recent firmware versions released in the meantime? Could it have been physically caused by the filament, for example some dust onto the sensor? In case the issue is still actual, do not hesitate to let me know info@prusa3d.com mentioning me ("attn: Alessandro"), we'll look into it. Closing the issue for now. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-24 by Prusa-Support