prusa3d/Prusa-Firmware #1732: Feature request: reset MMU2 from printer

langph, 2019-04-15:
When i a multi color print and a MMU error, somtimes I cannot correct the error: the mmu remains in a state of error. The only thing that helps is to reset the MMU. But by resetting the MMU from the MMU itself, it does not know which filament number it is on. So it thinks that the filament it was printing with before the reset is filament 1. If you can reset the mmu from the printer, it can recalibrate the mmu and make sure that it knows which filament is 1.

jsilva-iii, 2019-04-29:
I would also like to request this feature - on the MK2.5S there is no way to reset the MMU

GitSoks, 2019-06-25:
I would also love to see a feature where the mmu2 knows witch filament to load after a reset.
I would apriciate that a looooooot

gudnimg, 2023-07-04:
I think can close this issue after 3.13.0 is released. Some of the new MMU error screens give the option to restart the MMU in order to recover from certain errors.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg