prusa3d/Prusa-Firmware #1735: 3.7.0 - Nozzle temperature drop after first layer

francisdb, 2019-04-16:
The temperature drops from 215C to 200C while it's doing the outer perimeter

Movie showing the issue https://youtu.be/_sFY5GL0AWk

michalxfanta, 2019-04-16:
Hello, could you send us GCODE from video?

francisdb, 2019-04-16:
[Outer_Ring_v1_0.2mm_PLA_MK3S.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/3084805/Outer_Ring_v1_0.2mm_PLA_MK3S.gcode.txt)

leptun, 2019-04-16:
I am also having this problem. It is on a mk2.5s (r4 extruder). At first I thought it was because of the heater since it was only 35w when measuring resistance.

The actual problem was the print fan. On R4 it blows a lot more on the hotend than on R3. The heater is barely able to keep up when the fan is at full power. 

A silicone sock on the hotend will definately help, but this is a design issue and should not happen at all.

Also it wasn’t pid to blame since it climbed really slowly to the target during calibration with fan on

francisdb, 2019-04-16:
> A silicone sock on the hotend will definately help, but this is a design issue and should not happen at all.

I added that during the upgrade but my first prints failed so I removed it again, I guess the latest print fan shroud is not really compatible with the sock.

closed 2023-09-26 by github-actions[bot]