prusa3d/Prusa-Firmware #1741: Hotend should be raised higher after First Layer Calibration

selectnone, 2019-04-16:
The hotend is raised a few centimeters after First Layer Calibration is completed, but it doesn't give enough room to clean/inspect the hotend if the layer was a messy failure.

Could we please raise it to something more like ~10cm instead?

ochm, 2019-04-16:
I'm cleanig and ispecting nozzle after each print.
New print starts with Start Gcode as described below. No oozing during FLC.
...
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S160 ; set extruder temp before bed level
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 W ; go home all without mesh bed level
G80 ; mesh bed leveling
M109 S[first_layer_temperature] ; wait for extruder temp
M221 S100 ; Set flow to 100
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X50.0 E10.0  F800.0 ; intro thin line
G1 X100.0 E4.0  F800.0 ; intro thiner line
G92 E0.0
...

mws-rmain, 2021-05-10:
AMEN.
I'm having problems with filament curling back up & contaminating the nozzle during 'first layer calibration' routine. When finished, the printer moves the bed fully forward, but the hot end sits may 1 cm off the bed, while it asks if you'd like to repeat the process. PLEASE raise the hot end up to a respectable height so the nozzle can be easily cleaned in between calibration attempts. NOTE: I'm 6'2", and 240lbs, so 'respectable' means someone with 'normal' sized hands!

ochm, 2021-05-12:
> AMEN.
> I'm having problems with filament curling back up & contaminating the nozzle during 'first layer calibration' routine. When finished, the printer moves the bed fully forward, but the hot end [...]

It is not really necessary, just edit your start gcode to do MBL with nozzle at 160 or 170 dgC. No ozzing, no problem, not needed to move Z up.

mws-rmain, 2021-05-13:
The issue I'm referring to (same as @selectnone) is the 'First Layer Calibration' routine in the Prusa Firmware, NOT the bed-leveling routine (MBL?) that occurs at the start of every job. Modifying gcode is irrelevant in this case.

ochm, 2021-05-13:
You are right, my bad. 

> 
> 
> The issue I'm referring to (same as @selectnone) is the 'First Layer Calibration' routine in the Prusa Firmware, NOT the bed-leveling routine (MBL?) that occurs at the start of every job. Modifying gcode is irrelevant in this case.

You are right, my bad.

closed 2023-09-26 by github-actions[bot]