prusa3d/Prusa-Firmware #1749: Ideas for improvement (in Czech)

tkapler, 2019-04-18:
1.	Tiskárna by mohla schovávat příponu souboru, těch 5 znaků je strašně moc a pak to zbytečně roluje a je to úplně zbytečné (leda by měl někdo 2 soubory se stejným jménem jenom s jinou z podporovaných přípon, ale to nebude ani promile případů). Případně I třeba volitelně ty _MK3 co tam standardně přidává slic3r, za předpokladu, že tam není více souborů se stejným jménem

2.	Když tiskárna dotiskne, tak se přestane ukazovat název souboru – to je blbé, když se tiskne něco, co má víc dílů a člověk si pak musí někam poznamenávat, který je který.

3. třetí řádek na displeji po dokončení je zbytečný, ukazuje jen --:--. Mohlo by tam být vlevo délka spotřebovaného filamentu a vpravo celkový čas, tam kde je normálně v průběhu probíhající čas.  Jasný indikátor dokončení (tj. třeba "FINISHED") by mohl být třeba na tom druhém řádku na konci místo rychlosti, která je v tu chvíli už irelevantní 

3.	tohle jste teď myslím udělali: BTW mohla by být jiná ikonka u času podle toho, jestli má člověk vybraný čas dokončení nebo čas proběhlý (nevím jestli by to nešlo I nějak měnit přímo v té tiskárně).

4.	Případně by tam mohla být někde I ikonka ukazující, jestli má člověk vybraný stealth mode nebo ne (třeba u toho procenta nastavení rychlosti, ke kterému to nějak logicky patří)

ochm, 2019-04-18:
Chlape, dobry napady, mam par poznamek:
1. to mi prijde zbytecne, nastav si ve Slic3ru jaky ma byt format nazvu souboru, nicmene .gcode by to fakt nemuselo zobrazovat kdyz to stejne nic jineho nesezere.
2. souhlas
3. souhlas
4. nerozumim, remaining ukazuje za casem R, prepinani mezi ubehlym a zbyvajicim by se hodilo, kliden by se to mohlo na displayi po sekunde/dvou stridat, treba E jako elapsed, R jako remaining
5. jo to by se hodilo
Tak treba nas nekdo vyslysi...

michalxfanta, 2019-04-23:
Díky za nápady, zamyslíme se nad jejich implementací a snad se objeví v některém z příštích releasů.

closed 2023-09-26 by github-actions[bot]