prusa3d/Prusa-Firmware #1755: Feature request: Add option to unload filament at end of print

EngelmanJ, 2019-04-21:
Some woodfil filaments will burn if left in the hot end. Enabling 'unload at end of print' would solve that problem.

Similarly, if I'm printing in one material at 250C and the next print is at 215C, it might be nice to have it unloaded at the end of the print so there is no delay to change filament when I'm starting my next print.

drzejkopf, 2019-04-22:
Just add "M702" to filament end gcode...

EngelmanJ, 2019-04-23:
Before I posted, I wasn't able to find M702 in (https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/Marlin_main.cpp).

I looked again just now and found the code, but no comments. Then I realized I was looking at MK2 firmware. Better documented in the MK3 code.

drzejkopf, 2019-04-23:
@EngelmanJ You can also find that command in documentation (document is a bit outdated, but, as far as I know, all of these gcodes are supported by newest versions of Prusa Firmware) https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes#m702

closed 2023-09-26 by github-actions[bot]