prusa3d/Prusa-Firmware #1756: Tone of the sound doesn't change

crua9, 2019-04-22:
This is for 3.7.  I'm not sure if this is a problem with other recent firmware changes since it was a while since I updated my printer.

So I normally run the following code at the start of a print
```
; gcode start sound
M300 S440 P500
M300 S880 P500
M300 S440 P500
```

And the following at the end

```
; gcode end sound
M300 S440 P1500
M300 S587 P1500
M300 S440 P1500
M300 S587 P1500
```
The point of the sounds is to let me know when a print is about to start and end. The problem is the tone doesn't change even if the command above tells it to as seen.

JagerSprinkles, 2019-05-10:
Same as #1588

leptun, 2019-05-10:
This should be fixed in latetest Mk3 branch since timer2.h was disabled.
Not sure if it’s in this current release though. 

You might be able to try Mk3.7.0-RC1 and see the problem fixed.

crua9, 2019-05-16:
I can confirm this is fixed in the newest update.

Thanks devs.

closed 2019-05-21 by michalxfanta