prusa3d/Prusa-Firmware #1757: MK3S - MMU2S 3.7.0 - 1.0.

Albator91120, 2019-04-23:


Albator91120, 2019-04-23:
Hi, I noticed on tha last prints and potentially after a firmware upgrade that the extruder motor becomes very very hot. Seems that the temperature is so high that the filament can be bent around the bon,tech gear. Thanks to tell me if you have the same hot temperature problem and what is the solution

by-cx, 2019-04-24:
I have a similar issue on my brand new MK3S. It gets to 50°C while the room around is 28 °C. The extruder motor on my old MK2 is only 40 °C in the same conditions. Unfortunately, it melts PLA on bondtech gears after a few hours of printing - while printing more detailed parts.

m-cas, 2019-04-26:
Had the same problem with my MK3 that persisted after after the upgrade to MK3S. No way I could print PLA for longer than 4 hours. Extruder motor would get too hot to touch. I added a fan to the extruder, remixing a shroud I've found on thingiverse, and the extruder since then will get hot (but not scorching hot) and no "thermal runaway" anymore.
I power the fan from the same source as the hot end fan.
I think that PRUSA could easily add this as an option (the fan can be bought in the shop already and the Y cable to get the power is not going to cost a fortune).

Albator91120, 2019-04-26:
Did you have the same problem with a previous firmware version ? Do you have the MMU ?

by-cx, 2019-04-26:
@m-cas Thanks. I think I'll end up with the same solution. I have found something on Thingiverse too but the air was directed to the heat bed and that's something I would like to avoid. I think the problem is the bondtech gears are much closer to the motor in R4 extruder than in R3.

@Albator91120 Not sure if the question was for me but I don't use MMU and only tested the latest firmware.

Albator91120, 2019-04-26:
I found also the same : https://www.thingiverse.com/thing:2975582/ but I think that better is to find something to cool the bondtech gear instead of the external of the extruder motor. Perhaps I can adapt something identical with a part added in direction of the gear.

m-cas, 2019-04-26:
the thing yo mention is what I've remixed (it is now much lower, reduced about the height of the fan) but that was my personal aesthetic feeling  :D 
@Albator91120 : cooling the bondtech is curing the symptom as the source of the heat is the motor! It becomes so hot that the heat propagates to the gears.
Yes, I've had the problem with older FW version as well and I've worked with support as there was the strong suspicion of a defective extruder motor. No MMU.

Albator91120, 2019-04-26:
To cool the motor is ok but it's the external of the motor then a part of the heat can continue to go on the bondtech gear no ? but nice work

by-cx, 2019-04-26:
Based on this thread:

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-prusa-i3-kit-archive/increase-extruder-motor-current/

I think we could change these values in the firmware:

https://github.com/prusa3d/Prusa-Firmware/blob/166f8862036620dc86def1b1f16c5a9237d70c56/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h#L220

Unfortunately, there is not much documentation so it's trial and error approach. TMC2130_PWM_AMPL_Z is probably amplitude and that could decrease the amount of current running through the stepper motor. Even a small change could make a huge difference. Right now I am printing PETG for four hours and the stepper motor has 56 °C and bondtech gears match the same temperature. That's pretty high for PLA.

Another solution is this extruder:

https://www.bondtech.se/en/product/prusa-i3-mk2-5-mk3-extruder-upgrade/

But it doesn't look like it's compatible with the filament sensor in MK3S.

Also, I have received a new motor from Prusa. Really appreciate that. I will try it but I have disassembled the extruder so many times the last few days I don't want to do it again this weekend.

m-cas, 2019-04-26:
It could work but I’am not going to fiddle with the FW, I simply do not have the time to learn everything needed (and I used to do this kind of things professionally).
Before I couln’t print PLA longer than 4 hours without errors, the solution with the fan solved ALL my problems so I‘ll stick with it.
Prusa support was great as I‘ve got motor and extruder replaced (did all the work just once when I upgraded from MK3 to MK3S) but the extruder motor temp was still scorching hot. Fan installed, all ok. I am an happy camper. 
That is why I think Prusa should put the extruder fan ‚option‘ in the shop...

-
From my phone while mobile, please excuse any errors.

xoxota99, 2019-05-24:
I'm having this same issue since upgrading to the MMU2 / MK3S. Extruder stepper motor is getting too hot to touch.

TheWeindl, 2019-05-24:
I am getting temperatures of around 60°C on the extruder after 3 hours of print time on my new MK3S. I already installed a fan and fanshroud to the extruder, but this only gets me another hour or so of PLA print time, everything longer fails with extruder gear skipping due to melted PLA.
PETG works still fine for me, but the extruder is still scorching hot.
Anymore suggestions?

m-cas, 2019-05-24:
How did you configure your fan?
I have it in parallel to the hotend (sucking air up, not blowing down) and, with a slightly remixed shroud, I got rid of all temp problems (got 11+ hours of PLA printing, before 3.5 was the maximum).
The fan is the same noctua 5v that is used on the left to cool the hotend. 

-
From my phone while mobile, please excuse any errors.

xoxota99, 2019-05-24:
Anyone have an example of a fan for the actual extruder motor? (As opposed to the hotend or the heat block). I'm hainvg trouble visualizing your solutions...

Albator91120, 2019-05-25:
[image]

here is what I did. I use the first fan replaced when I upgraded my MK2S. I added a pipe then a small part of the air flow is directed on the bondtech gear. I can give the stl file if someone wants to modify the original file. At this moment no longer hot problem.

m-cas, 2019-05-25:
I've bought a noctua (same as the one prusa uses on the hot-end) or buy it directly from prusa as replacement part. See attached pic for setup. The only complicated thing was building the splitter cable as I didn't have the pieces immediately available.
[image]
You can see the fan cable piggybacked on the cable bundle as I didn't want to disassemble the whole thing. Next time I do an upgrade I will put the cable in the bundle. The STL file for the shroud is available in case someone wants it.

xoxota99, 2019-05-26:
@m-cas Can you share your STL file? This might be just what I'm looking for. I assume you printed in PETG?

m-cas, 2019-05-27:
sure, here you go (it is a remix from a shroud on thingiverse but I’ve made it less bulky):

https://www.dropbox.com/s/qezx0esyj234j7s/cooler%20down.stl?dl=0
https://www.dropbox.com/s/gsfr2nn8cgpmoxe/mac%20cooler%20up.stl?dl=0

The upper part is useful if you have “too much cable” giving you two channels to wrap it around the body of the fan. If you go for putting the cable in the same channel as the hotend fan, than you may just use the lower part.

I’ve printed it in PETG and fixed it with a dual-side tape that is used to fix heatsinks to transistors (thermal resistance).

I’ve bought a standard noctua fan from a local shop, if you go for the prusa “hotend replacement fan” you get all you need including the long cable and you save yourself a connector ;)

Cheers

Qernek, 2019-05-28:
I have the same problem since an MK3S / MMU2S was updated. The engine gets so hot that it doubles the PLA and jams the bondtech gears

nunofgs, 2019-05-28:
I've had the same issue since upgrading to MK3S and 3.7.0. Prior to that I printed for months on end with no issue.

A printer that used to be perfect and super reliable has now become a constant source of frustration and disappointment. I can *sometimes* get a <2h print to finish if it's not too hot outside, at night. (read: under 22º C)

I have am infrared "Thermal Gun" and the extruder motor goes up to ~60º C when the PLA becomes too maleable and slips on the gears causing it to stop extruding.

The Prusa3D facebook group also has a bunch of people complaining of the same thing. This thing *has* to be more widespread than it seems...

Has anyone found a solution that doesn't involve adding a fan? I can't believe that a printer that worked well for months now requires modding to function properly.

m-cas, 2019-05-28:
I had the issue even before, I’ve started with a MK3 and from the beginning PLA after 3.5hrs would fail. Always. I took the upgrade to MK3S as chance to mod it and it’s gone.
Prusa support helped me very much by sending new parts as they suspected a bad motor.
I can only say that the fan seems to radically change the behaviour in positive *for me*. I understand your point if view though and I agree that it seems to widespread.

-
From my phone while mobile, please excuse any errors.

xoxota99, 2019-05-29:
Thanks for the STL file, @m-cas. Unfortunately this design doesn't work with the "new" (MMU2S) extruder design. I'll have to remix it or something...

DominiqueFuchs, 2019-06-02:
Same here, unfortunately. Upgraded to 3.7.0 one week ago (MK3S) and experienced full clogging through the PTFE tube to the bondgears while doing a 6h print PETG - yes, PETG. After disassembling the relevant parts I‘m suspecting heat creep, now after reading this issue I‘m wondering if the motor itself is the cause. Done about 120h of this kind of prints before without clogging.

m-cas, 2019-06-02:
@Dominique, I recommend you to contact Prusa support. They were very helpful!

-
From my phone while mobile, please excuse any errors.

pegelf, 2019-06-07:
I've been doing the same 14 hour print ten times and since a few weeks and I can't repeat it anymore, because on the half, the extruder doesn't move the filament anymore.

Instead, it is clicking as if the extruder were blocked and the motor gets very hot. After canceling the print and cooling up i can restart it and there is no problem for a few hours and then the same problem happens again. So I can't finish my project in time.

mkreider, 2019-07-01:
@m-cas

I wonder where you got the helpful support _sigh_ 
It seemed to me they were briefed to blame the problem on everything else. Had 5 chat sessions with support about this so far. They all claimed that its perfectly fine, 'the motor is good up to 100 °C". Then followed a very long list of things to blame for the blocking and try, from the ambient temperature, PTFE tube, partially blocked nozzle, thermistor, fan, thermal paste ... I've had the hotend apart 7 times to show good will and even ordered a complete new hotend because I was getting sick of it.
Printer still near useless since the upgrade to MK3S with MMU2S.

There was only one tech support guy who had a slip of the tongue and then hastily tried to take back his statement:

Douglas 01:54:11 pm
what is the room temperature where you are printing please ?

Mathias Kreider 01:55:17 pm
Right now between 21 and 26 degrees, depending on daytime.

Douglas 01:58:38 pm
as we are getting customers with this issue needing to move their printers to a more constant cool area for printing with no problems

Mathias Kreider 02:01:15 pm
Let me get this straight: It worked fine when it was a standard MK3, now after the upgraded, I need air conditioning installed to run the printer during the summer months ? Please tell me you are joking

Douglas 02:01:43 pm
yes we are even adding more thermal paste to our hotends lately in this regard but was not suggesting your issue was exactly the same was just saying this is also a cause, did not mean to suggest anything sorry. have you checked the extruder pulley to make sure it is perfectly strait to make sure the filament goes strait into the tube

m-cas, 2019-07-01:
that the support guys go through some long checklist seems to be the industry standard but my experience in multiple sessions with Shane (he works on a US time schedule so I contacted support after 6PM CET and asked for him) were excellent…
I do not have the MMU2S though so it could be that this piece makes a difference...

mkreider, 2019-07-01:
I don't think it does make a difference. Ironically, using the MMU is the only thing that works right now because frequent filament changes mean a cooling period for all types not currently used. 

I tried printing with disconnected MMU and directly feeding the filament from a spool to the extruder because I was suspecting the higher resistance by pulling filament through all that bowden tubes and this buffer thing would heat the motor excessively, but that was not the case. The difference is about 2-3 degrees for my setup. Personally, by now I think its simply a firmware problem by setting a ridiculously high motor current. I will reinstall my previous FW for testing and then spend some time with their git repo to find the commit(s) after which it all went to hell.
I'll post my results when I get something solid.

m-cas, 2019-07-01:
please do!!

by-cx, 2019-07-01:
Since firmware 3.7.1 my MK3S got better, 50 °C on extruder motor, sometimes little higher, but still pretty close to the point when PLA is turning into a "flexible" filament. As a comparison, MK2's motor is 40 °C. There is new firmware 3.7.2 I am going to install soon, maybe it's better with that one.

I refused to use a fan and I think the only way from this right now is either an upgrade from Prusa (firmware or hardware) or to buy this baby:

https://www.bondtech.se/en/product/prusa-i3-mk2-5-mk3-extruder-upgrade/

It has gear ratio 3:1 between the extruder motor and Bondtech gears, that means less stress on the extruder motor and also longer way to 

I workarounded the issue by buying powdered sheet and I print PETG only.

I wanted to buy MMU2 upgrade but until this is fixed I don't want to deal with extra issues.

xoxota99, 2019-07-01:
I had problems printing with Prusa filament (PLA) because of the extruder motor / bondtech getting too hot (60C after a couple hours printing). Switched to Hatchbox, problems are gone.

I'm still worried that this high temperature is a symptom of something that's going to put more strain on the motor though. My chats with support were pretty similar to the above, blaming everything from ambient room temperature to thermal paste.

Once I explained that I bought the hotend from Prusa less than a month ago, fully assembled, that theory went out the window.

mkreider, 2019-07-02:
So, I was trying to find out if the sudden heat is firmware related or due to the MK3S extruder design. The short answer is: No, it's not the firmware.

I downgraded the MK3S hardware setup yesterday to MK3 FW binaries that had definitely worked on my MK3 printer without the extruder stepper going for the burn. The main difference for the firmware is the kind of filament sensor, but since the printer also runs with the sensor unplugged, that was not a problem and soon the MK3S hardware believed it was a MK3 again. 

Unfortunately, it turned out the batteries for IR thermometer were drained, so I had to go for the good old fingertip-comparison yesterday: Hotter or colder than the 60° heatbed ?
My test was printing benchy for half an hour,  I took the gcode from the prusa demo files.
Used firmwares were MK3 3.4.1 / MK3 3.5.2 / MK3 3.7.2 / MK3S 3.7.2

For all versions, also the known 'good' ones like 3.4.1, after half an hour the extruder stepper felt hotter than the heatbed, ie. getting painful at the fingertips after about 5 seconds and was thus >= 60°.

I then tried full factory resets and new calibration, because I was suspicious of the data still in the eeprom. However, he result stayed the same, motor case was hotter than the heatbed. Finally, I flashed MK3S FW 3.7.2, did another wipe and calibration, with the same result. So I could not confirm my original hypothesis that the extruder current had been increased for the MK3S FW > 3.4.

I then went through the git repo and tried to establish a timeline of commits relevant to the extruder steps or current values. Here's just a few of them:

2018-05-02 964d019dd18 (Pre 3.2.0) MK3 - extruder current = 30
2017-08-30 f563618b1c0 (Pre 3.1.1) PINDA thermistor + extruder current adjust
2017-06-06 3fdb630c42f7 (Pre 3.1.1) ... Increased currents in Z and E axis ...  E axis microsteps changed to 64

This is consistent with the test - there are changes to the extruder current (quite hefty ones in fact, especially the one from 2017-06-06 increases run currents by factor 1.5 and hold currents by factor ~ 3.5), but - and this is the important bit - they all predate my 'good' FWs, ie. are before v3.4.1. 

So the preliminary conclusion is: The firmware is innocent and the MK3S mechanical design seems to facilitate the heat buildup in the extruder motor. Full conclusion would be to build back to MK3 extruder and install 3.4.1 which should be colder than everything I tried yesterday, but it will be some days before I will have the time to do that.

m-cas, 2019-07-02:
> 2018-05-02 964d019 (Pre 3.2.0) MK3 - extruder current = 30
> 2017-08-30 f563618 (Pre 3.1.1) PINDA thermistor + extruder current adjust
> 2017-06-06 3fdb630 (Pre 3.1.1) ... Increased currents in Z and E axis ... E axis microsteps changed to 64

@Mathias, I had the problem with my MK3 from the very beginning (Dec 2018).
I upgraded to mk3s (and swapped the extruder and stepped motor with warranty replacements from Prusa), added the small fan on top of the extruder motor and the problem went away (didn’t have the time to do piece by piece).
My printer is in an enclosure (IKEA), and looking at the forum it seems that the major difference is adding the fan as a number of mk3s customers seem to have this thermal issue.

mkreider, 2019-07-02:
@m-cas  
> ... I had the problem with my MK3 from the very beginning (Dec 2018). I upgraded to mk3s (and swapped the extruder and stepped motor with warranty replacements from Prusa), added the small fan on [...]

Ok, thanks for sharing, that's interesting. In that case I'll do a quick first test with a sticky thermal pad and cooling fins, measure the effect (memo to myself, buy batteries for IR thermometer), add a fan, measure again and then perhaps see if more cooling fins on the other sides and/or a shroud have a meaningful impact. 
I actually like the idea of complete passive cooling by adding cooling fins to all four exposed sides and let the movement of the extruder head do the job of moving the air.
I'll post the results once i've got it done

m-cas, 2019-07-02:
@mathias,
Fyi, I did test with a passive cooler on top (with a thermal sticky pad) but it didn‘t change anything.
I‘ve modded a shroud to keep it low profile and installed the fan in ‚sucking‘ mode to avoid blowing air on the print surface.

-
From my phone while mobile, please excuse any errors.

xoxota99, 2019-07-02:
One data point: My motor is really only getting up to 60C, even after twenty hours of printing. Seems to plateau there. This should really be hot enough to soften the PLA filament.

But then again, I don't know how hot the actual bondtech gear is getting. Only the motor casing.

mkreider, 2019-07-03:
@c-mas
My plan was to add several heatsinks and measure the effect for each subsequent one with and without active cooling. Maybe it is already enough if I make the motor look like a cube shaped hedgehog. If that gets me ~ 6-7°, that would be more than enough. But if it's going to be active cooling, I'd like heatsink(s) plus shroud for maximum effect.
Moving the air upwards is a nice touch! I'm just wondering ... the air will flow in from somewhere, did you ever test with something sensitive like ABS to see if there is a flow of colder air going across the bed towards the extruder or did you just use PLA?

@xoxota99 Thanks. Mine is around 63 - 65° since we entered the hot phase of the summer and stays there. Measuring the effect is on my test todo list for the weekend, I was planning to pause mid print and open the flap to measure directly on the filament.

m-cas, 2019-07-03:
@mathias
I’ve never printed with ABS, I us only PLA and PETG. My problems were exclusively with PLA.
Using a shroud will leave only place for an heat sink in the front, unless you want to end up with an extruder that is HUGE :D (depending on which material you choose for the shroud and heatsinks, consider the added weight of the whole on the belt)

mkreider, 2019-07-10:
@m-cas I finally got around to trying passive cooling. Ambient temperature was 26.5 when I did the uncooled reference and 27.6 when I did the test, so the weather was slightly biased towards the reference. I've attached four 40x40x11mm heatsinks to the accessible surfaces of the extruder stepper with 0.5 mm arctic heatpads.

My measurements were taken after 30 minutes and after 5 hrs print time respectively.
Bare: 48.0°, 56.8°
Passive Cooling: 39.5°, 42.8°

So with the four heatsinks, I get a difference of ~ 14 ° at the plateau by adding a total weight of 60g, which is lighter than the noctua plus shroud. No more jamming trouble so far neither, I'll continue testing. So for my setup, this works great at minimum effort and weight, but I am not using an enclosure. Active cooling is probably back on the table if I build one.

Cheers - Mat

m-cas, 2019-07-10:
Cool (pun inteded), happy it worked out well! Could you share a pic?

-
From my phone while mobile, please excuse any errors.

mkreider, 2019-07-11:
Here you go
[image]
[image]

Ghukek, 2019-07-14:
Thank you for this discussion. Unfortunately my wife won't allow me to run the printer in our air conditioned bedroom so the summer temperatures are sitting in the eighties (high twenties for the rest of the world).  I'm getting regular jams due to the extruder running hot. Would you mind sharing the heatsinks and thermal connections you are using @mkreider ?

mkreider, 2019-07-15:
@Ghukek Sure, I ordered these. Don't know if they deliver outside Germany though

If you can't get these, I don't think the make of the heatsinks will matter much as long as the dimensions fit. As for the pads, I chose one with good thermal conductivity (> 5W/mK) and where customer reviews said it sticks very well (so there's low risk of them being shaken off the print head).  Thickness of course as thin as possible, I found that the 0.5mm is quite sufficient to cover the grooves on the NEMA 17 housing. Probably needless to say it, but make sure both housing and heatsinks get a very thorough alcohol scrubbing with a lint free cloth before applying the pads

Heatsinks
https://www.amazon.de/gp/product/B07C7SJHTH/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&language=en_GB&psc=1

Thermal pads
https://www.amazon.de/gp/product/B00UYTTLI4/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

cggorman, 2019-07-15:
I looked into this a month or so ago.  A single heatsink on the motor with a Noctua attached keeps the motor very cool. It is critical that the Noctua blows TOWARD the heatsink.

Passive after an hour of printing with a room temp of 25c. (42c motor)
https://i.postimg.cc/G3z572G0/flir-20190601-T110846.jpg

Active in identical conditions. (33c motor)
https://i.postimg.cc/HsxPfg9r/flir-20190601-T140019.jpg

Master3DPrints, 2019-07-15:
Are any of you finding this to be a reliable and permanent fix? I'm thinking to do the same with a fan and shroud or heatsinks. 

Also, I spoke with Prusa. They're working on a FW fix. Here is a user customized FW file sent to me from a Prusa representative through their online chat. It takes you back to the old days of 3.7.0. lol, But if you rely on your printers as I do I guess it's worth it. Can anyone verify if it is 100%safe and secure. I'm always a little paranoid when downloading any unverified file from the web. https://cdn.livechat-static.com/api/file/v2/lc/att-old/9644675/14810561dd48faa6bfce47731804d664/MK3S_3.7.0_lowtemp.zip

[image]

m-cas, 2019-07-15:
Fan and shroud did the work for me (meaning: no meltdowns when printing PLA for longer than 2.5/3 hrs).
Did you got info on what was changed in the custom fw to keep the temp low?

-
From my phone while mobile, please excuse any errors.

TheDuggem, 2019-07-15:
Here is a U.S. Amazon link to the ones I used: https://www.amazon.com/gp/product/B01HSBM72W

I put 3 of them on a couple of weeks after getting my printer and have been printing successfully for over a year now.  They're held in place with a narrow zip tie for easy removal.  No fan + extra wiring needed.

mkreider, 2019-07-16:
> I spoke with Prusa. They're working on a FW fix. They gave me a user customized FW upgrade.

Hmm, there's currently no commit for it in the main (MK3) branch and no feature branch for it yet. I'll definetly  have a look once its out, want to know what they did exactly.

Master3DPrints, 2019-07-18:
> Did you got info on what was changed in the custom fw to keep the temp low?

I didn't. They just mentioned that it helped keep the motor from getting too hot. It helped but then I had numerous "Bed Thermal Runaway" warnings which stopped prints.

Master3DPrints, 2019-07-18:
> Here is a U.S. Amazon link 

Thanks! purchased them a few days ago and will install when they arrive. I updated the FW to 3.7.2 and now the fan and shroud aren't sufficient to keep the temp down. I didn't have a failed print but the temps went up for sure. I had to put a normal desk fan on the printer to keep the temps down.

cggorman, 2019-07-19:
Funny how different everybody's use caes are.  I'm trying to keep the motors below 100c in an 80c heated chamber!

leptun, 2019-07-19:
@cggorman Wow... That is really hot. What are you printing in that chamber? PEEK? :)

cggorman, 2019-07-19:
Large CF-PC parts. Would like to try PEEK but need a Mordor plate which basically means a new printer.  That's next...

TigerClaw33, 2019-07-30:
Hello, 
I am using a Prusa i3 MK3s Zaribo MOD. After the MK3->MK3s Upgrade and the newest firmware from May/June 19 I got a clogged nozzle with a new PLA. I thought it's because this new material. I also thought it's because my enclosure and it's to hot inside. But yesterday I verify the extruder temp. The extruder is very very hot. So I tried to reduce the tension of the idler ... It's not really better but it works at the moment. 

But I think I have the same problem

Master3DPrints, 2019-08-01:
> But I think I have the same problem

Yeah, you most likely do. Try ordering this  https://www.amazon.com/gp/product/B00NEMGCIA/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1
from amazon and printing a fan shroud like this 
https://www.thingiverse.com/thing:2975582
from Thingiverse. Takes some work but it solved my problem on both my printers.

unixunion, 2019-08-12:
I just noticed my extruder motor is around 55 degrees celcius. This is far too hot IMO.  Is Prusa gonna fix this?

mkreider, 2019-08-12:
Two tech supporters told me in July that 'some customers' were experiencing this and Prusa is aware of it, but they were very vague if Prusa is going to include a fix in the next release.

mkreider, 2019-08-12:
@TigerClaw33

Yes, you probably should cool the extruder. For me, it worked very well. However, for the sake of completeness, there is one more possible cause for jams that I recently experienced. 

A bad thermistor in the hotend can lead to jamming without the firmware seeing anything unusual.

Sounds odd? Here's the story:
I ordered a new hotend from Prusa when huntning down the jamming related to heat from the extruder motor. Cooling the motor fixed the issue in the end, but the new hotend's thermistor was acting odd. It reliably jumped by about 20 degrees when passing the 60something degree mark, which was causing temperature runaway issues in firmware. 

So I swapped the thermistor for a brand new one from Prusa. After that, there were no firmware errors anymore, but I suddenly had jamming issues again when printing PLA. Placing the probe of a digital thermometer into the heater block from the other side, it turned out the replacement **thermistor was showing 14-22 degrees more than there actually were**. And at a ~180°, PLA apparently can put up too much resistance when forcing it through the nozzle.
To prove the point, I tested extensively with a +20° offset in octoprint, and behold, all troubles were gone. 

Since I had received not one but two bad thermistors from Prusa, quality control might be a bit on the low side here. I'd definitely recommend testing the sensor for everyone who suffers from persistent jamming trouble when using PLA.

In their defense (or rather in defense of customer support), they sent me two new thermistors and a little something for my pains when I complained. The new sensor fixed the issue as expected.

mkbel, 2019-08-16:
@Albator91120 Can you check your gcode, if it doesn't contain some current manipulating commands like M907 M208 M911 M912 ?
Can you post your problematic gcode here?
Older version of Slic3r is know to manipulate (incorectly) extruder motor current. This extruder motor current rise during filament change is unwanted relict from MMUv1, where it made sense.

ThePlox, 2019-09-04:
Hey guys , I encounter this problem on my Prusa I3MK3S in summer months. I fix it like mkreider did.
Prusa support said this issue is random and there might be a problem with my other extruder parts which is not. The heatsinks works well and extruding didnt stop on my 50 hours of printing. May I ask where do you live ? In my city the temperature goes up to 40 C in the summer months. 
I note that only pla softens in the extruder gears.

[81 more replies]

closed 2022-05-20 by Prusa-Support