prusa3d/Prusa-Firmware #1764: Enhancement: Make the tuned settings 'sticky' when changing filaments.

esoren, 2019-04-24:
This is similar to issues #1364 and #1208, but not the same since it includes the full set of settings and not just speed.

Since adding the MMU2S the 'tune' menu of the printer is basically useless since all of the settings are reset on every filament transition. An important use-case for this is tuning the temperature on-the-fly to get acceptable tips for challenging materials like BVOH.

meh, 2019-09-03:
Aight, just experienced this and I'm fed up with it, so I'm gonna fix it myself.

Is it a solution for you as well if the settings just stick, but you can't preemptively change settings for each slot? Because that'd be much harder.

kdsohn, 2019-09-03:
For me that would be fine enough if I could only read out the settings after the print in any way to include them into the project.

meh, 2019-09-03:
@kdsohn that would be harder and might be better as a patch to prusa-slicer instead, but I'd rather not touch that.

kdsohn, 2019-09-03:
It did not mean an automatic transfer, it would only be handy, if I could see all modifications after the print (and before a new print resets everything).

meh, 2019-09-03:
@kdsohn no I get you, but that's an entirely new LCD view, which would be harder than what I'm doing :panda_face:

kdsohn, 2019-09-03:
I understand. But as has been said before, the total tuning feature as it is nearly useless with the MMU, so it seams really worth a major update.

meh, 2019-09-03:
Oh absolutely.

kdsohn, 2019-09-03:
Perhaps the changes are not too hard, if you could reuse the filament selection for MMU single prints before showing the tuning menue. This can be a feature request.

closed 2023-09-26 by github-actions[bot]