prusa3d/Prusa-Firmware #1769: failed to Load MMU

morimorida, 2019-04-26:
So this happened to me twice now since I upgrade my MK3 to MK3S and MMU2S with newest 3.7.0 firmware and 1.0.5 MMU2 firmware.

1st attempt:
After printing for over 12 hours+ and waking up in the morning, noticed that the print stopped.  LCD screen says MMU failed to load.  Fixed the filament issue on back of MMU, however, there is no option to resume. I manually fed the filament to the top of the extruder then suddenly it started to move but it was trying to print inch off the wipe tower without nozzle being heated up automatically. Gave up and reset printer.

2nd Attempt:
After printing over 12 hours+ (different height where it happened though), noticed that the print stopped. LCD screen says MMU failed to load.  Fixed the filament issue on back of MMU, however, the LED on MMU is blinking green/orange alternately but again on LCD menu there is no option to resume print or does not automatically resume print.  This time around, I hit the reset button on MMU.  nozzle start to move and at first started to print inch off the current print layer location.  However, the nozzle was not heated up nor MMU did not feed the filament.  After it tried to print at actual model, it did dive down to the supposedly where it left off prior to failure to load.  Again, since nozzle was not heated up automatically or filament not fed automatically, nothing was printing. Gave up and reset printer.

On both occasions I did use Octoprint.  Noticed that the current 3.7.0 firmware have problem with Octoprint, but the bug says that it occurs on first layer?

"known bug resulting in failed print after first load filament error."

I'll try to print again tonight, same print but with SD card and see.

mvasilakis, 2019-05-02:
I can confirm that I've experienced similar printing with Octoprint as well as printing from SD.

mkreider, 2019-06-19:
Can confirm this 2x with FW 3.7.1, MMU FW 1.0.5 and Octoprint. I think I also saw it once from SD, but I am unsure here. 
Symptoms are that  LCD says MMU load failed, MMU does react to button presses as if it was idle and Terminal says Recv: echo:busy: paused for user.

I found a workaround for me which saved my prints:
After I pressed the MMU middle button to verify the load, I can get printer back to life by simply pressing the knob on the printer LCD once. The printer then says just 'Heating...' and once its done, it will resume print.

I tried sending it a M602 code to resume and also a T? to select my filament for load when its stuck in that bug. At first it looks like these commands are ignored, but this is not the case. The execution of all code is blocked until the button on the panel is pressed and temperature is fully restored. After that it executed all inserted gcodes in order.

skylord73, 2019-08-04:
same problem.
I tried in Single filament
The MMU loads filament fine if I cut it 45°, but after first print, when it tries to load filament again, the filament arrives the extruder, triggered the IR, but for some reason, the printer unloads it, and tries again, without success. I have to eject filament, cut it again at 45° then it's loaded fine again.

closed 2023-09-26 by github-actions[bot]