prusa3d/Prusa-Firmware #1770: MK3S + MMU2S - FINDA stays in 1 state halfway through print

n8maher, 2019-04-26:
The printer works fine for half a print, then the FINDA sensor value becomes locked at 1.  Resetting the MMU does not have any effect, nor does unplugging the MMU while the machine is powered up.  Only turning off the printer and turning it back on again fixes the issue.  It has happened two times in a row.

rlobbins, 2019-04-30:
I also have this issue with my mmu2/mk3s, I am running 3.7.0 (with octoprint fix), and mmu firmware 1.0.5. I will try to get a recording of it.

harvdog92, 2019-04-30:
This happens for me too. The FINDA works when there is no print running (I test it beforehand with filament). Then, when I start a print it loads the first filament and the value changes to 1 and it never changes back. It stays at 1 for most of the print, even when the selector is moving to change filaments. When the printer senses an issue with loading and unloading is when it messes up a print. It will try to reload a filament but then will just keep it in the MMU, not moving it past the selector thinking there is something already in there.

Last night it started just not loading filament at all after a factory reset. It will load for a few centimeters then retract filament.

n8maher, 2019-05-06:
Developers:  let me know if you need more detailed bug reporting, and I'll see what I can do.  I think every print has had at least one load error that needed human intervention before the FINDA sensor value became locked.

jnsanders1983, 2019-05-08:
I have found that the state of the FINDA does not change during the load procedure using the buttons on the MMU, and also during tool changes. Which leads to an issues where filament will be completely unloaded causing the MMU to error on unload. It’s as if it fails to poll the sensor state while the machine is performing a task. Thought when the machine is idle you can see the state change when inserting and removing filament at the sensor. Technical support denies that there is a firmware bug and says its the FINDA sensor. They sent me a new one I will received today and I plan to install it and retest. This time I will record with a GoPro for evidence. I will also try to replicate it many times to see if there is a pattern. I found that from start of a print, I can print for about 26 min before I get an unload error. I want to say that even though the status of the FINDA does not change during some operations it must work in the background some how, but in some specific instance, root cause unknown, the state is read as it is on the screen and therefore causes an error where the machine believes the filament is not unloaded, which eventually causes the MMU to believe an error occurred and pause the print and wait for user attention.

Ghostphart, 2019-05-10:
Similar issue here. The Finda probe LED works as intended, but the printer does not detect the filament, the UI shows 0 for the filament sensor, even when the Finda probe LED turns off. I have checked the cables to make sure they are seated, and the other sensors are working fine. 

I have had to take the Finda probe out and replace it many times due to stringing of the filaments, which caused the Finda probe to stop working and filament loads to fail. Take the probe out, a quick shot of compressed air, and then reseat the probe and make sure it is working with a test bit of filament.

cbuhl79, 2019-05-15:
I'm having the same issue. If I watch the Sensor Info menu while it's swapping filament, the FINDA setting is always at 1. When the print stops for MMU attention, it still stays at 1, even when there's clearly no filament in place. Sometimes this hasn't caused any problems, and I'm able to resume the print after clearing an issue. But other times, it completely kills the print. The MMU can't ever recover properly (it keeps trying to unload, presumably because it thinks it's still loaded), and only powering down the entire printer gets it to reset.

Note that when this is happening, even removing or adjusting the FINDA probe doesn't make a difference. It's still always stuck at 1.

I've also had the opposite happen, where at some point it gets stuck at 0, and won't change back to 1 until the printer is reset.

This doesn't appear to be an actual problem with the sensor itself, since everything works properly immediately after a reset.

Ghostphart, 2019-05-15:
Seems that 3.7.1 firmware is out for the MK3 now. And while there is still weird behavior for the Finda probe, it does work to print objects. I still have to cut filament tips in the MMU2S, but it is working for the moment.

cbuhl79, 2019-05-15:
@Ghostphart did 3.7.1 have any fixes related to the FINDA? I didn't see any in the changelog notes. I've been able to print succesfully on 3.7.0, even with the FINDA problem. It's just that if it pauses and needs attention, there's a chance it completely fails, which I *think* is related to the FINDA problem.

Ghostphart, 2019-05-15:
@cbuhl79 Not anything that I could see, but perhaps it was some conflict that was resolved, or maybe better behavior for the sensors.
I am trying another Gustav the turtle multi color prints, and have lowered the temps of my Esun pla to 205 for layers above layer 1. Hope it will fix the stringing left behind in the MMU unit.

AndrewR15, 2019-05-31:
Just got my MK3s working perfectly then  went to add the MMU2... Just like your the FINDA sensor displays 0 When filiment is loading (usually says 1 about 10 seconds later) often causes errors. another problem that might be related is that my Idler sometimes after swappign colors will stay engaged forall the printing of that color(without the MMU2 feeder spinning). This causes it to completely skip the color for that section. print sometimes finished but with large gaps.

cbuhl79, 2019-06-01:
I think that my main problem was with the IR sensor not working properly, and the FINDA sensor might have just been a red herring. 

I read somewhere else that some people were having problems with the IR sensor not detecting the lever properly. Apparently the black filament actually still lets some IR light through. One person recommended putting whiteout on the tip to block the IR light. I did this, and I'm no longer seeing this issue.

andrepruitt, 2019-06-03:
I’m having the same problem. It’s happened twice so far while printing from the SD card. I’m using the MK3S with the MMU2S and the latest firmware. I set the Slic3r to: “0.15mm SOLUBLE INTERFACE”.  

Both prints worked fine until the FINDA sensor got stuck to a value of 1. Pulling the sensor out of the MMU2 had no effect on its value. Cycling power on the printer restored the FINDA.

sebleedelisle, 2019-06-12:
I have this exact problem as described by @andrepruitt - FINDA gets stuck on a setting of 1 and doesn't reset even if I pull it out. Power cycle resets it.

blu3lit3, 2019-06-15:
Also an issue for me, FINDA gets stuck on. Could this be a heat issue? I have an enclosure on my setup and like @jnsanders1983 mentioned, its like a half hour into the print before it fails me.

sebleedelisle, 2019-06-15:
Hi everyone, I think that the FINDA being stuck on 1 could be a red herring. My theory is that the MMU2S electronics stops transmitting the FINDA state to the main printer board when it goes into any of its error states. So even if it turns off, the new state doesn't get reported in the main LCD display.*

Personally, my issue seemed to be with a VERY over-sensitive IR sensor, once I got that better aligned (and loosened the spring screw on the main extruder) my issues seemed to be mostly fixed. 

* perhaps someone from Prusa could verify this?

frankracis, 2019-08-29:
There's definitely a problem here, and it's not the hardware.
I was trying to load some flex filament for a single-color print.   This went wrong many times in many different ways.  At least 2 times after a failed load:
* The FINDA LED was working correctly
* The FINDA status on the LCD was stuck at 1
* Restarting via the front panel X button restored proper operation

meh, 2019-09-21:
This happened to me as well, lost a quite complex print :(

tmcdermaid, 2019-10-21:
Having the same problem as everyone above and in numerous other forums. The F.I.N.D.A. probe sensor state registers 1 on the first film ant load and remains as 1 when the MMU2S attempts to change filament. Some people have said the IR Sensor is causing the filament load failure but I would disagree...as long as you have calibrated and verified the IR Sensor. I believe this is directly related to the F.I.N.D.A. Probe and Firmware. There are too many posts for this to be an “other” issue. While watching the Sensor State on the LCD Screen, the F.I.N.D.A. Remains 1 as the MMU2S and Extruder initiate a filament change. As the next filament is loaded to the Extruder, the IR Sensor changes to 1 but the Extruder quickly reverses the filament so the IR Sensor changes to zero as the filament is unloaded then loaded as a second attempt. F.I.N.D.A. sensor remains a constant 1 throughout. As the filament starts to load again at the Extruder, the IR Sensor changes to 1 then the Extruder changed direction and the filament is unloaded as the IR Sensor shows zero. After two or three attempts, the filament load failure occurs. I’m not understanding two issues which I hope someone can help clarify and possibly resolve: first, I don’t understand why the F.I.N.D.A. Probe Sensor state remains 1 regardless of filament being present or not. Second, during a filament change when the filament is pushed down to the Extruder and my IR Sensor State correctly reflects zero to 1 as the filament arrives the Extruder gears, why does the Extruder reverse direction immediately after the IR Sensor changes to 1. My only thought that ties these two issues together is, during a filament change the F.I.N.D.A. Probe should change from 1 to zero as filament is unloaded then back to 1 as the next filament is loaded at the MMU2S then pushed to the Extruder where the IR Sensor is triggered from zero to 1 as the filament passes through the gears. If, the F.I.N.D.A. Sensor fails to change from 1 to zero (during filament unload) then back to 1 as the next filament is loaded, wouldn’t the Firmware check the sensor state which is saying the filament never unloaded (since the sensor state remains a constant 1) so even though the IR Sensor correctly reported the filament arrived at the Extruder, the Extruder immediately reverses direction to attempt the unload/filament change since the F.I.N.D.A. Never reported a zero?  I know this is wordy but since it continues to happen to o many users, I’m in ope of filling in any holes. As a side note, the F.I.N.D.A. Probe has been removed, cleaned, the hole cleaned, the steel ball cleaned, etc. to no avail. Thoughts?

someguy4253, 2019-11-15:
Like to chime in on this as well... I too have this same problem were the MMU2S Finda is getting stuck on 1 when no filament is loaded and this has occurred with a 0 as well when the filament is loaded. 
* Updated the MK3S firmware to 3.8.1 last night (not sure if that's the culprit since I have just heard about this bug a few days ago) I noticed something new started happening ... it was not getting stuck per-say,, it was very delayed in the transition from unloaded 0 to Loaded 1, (about 2+ second delay) and then the printer ejected the filament with a MMU error, this occurred many times and could never load the filament again and had to shut down the printer to clear it.

Hope this gets fixed soon since my MMU is worthless to me If you cant complete a print.!!!!

spiff72, 2020-01-12:
I just found this issue from a google search after I happened to look at the Sensor Info screen during a long print.  

The print is printing perfectly, filament changes are running fine, FINDA LED is changing states as it should during the changes, but the Sensor Info screen continually shows FINDA state as "1".

JohnSvensk, 2020-02-21:
This is NUTS! Doesn't anyone have any ideas? Ten months, and no improvement?! 
What is this? I wish my C++ skillz were sufficient to debug the firmware, but I'm afraid they aren't...

spiff72, 2020-02-21:
I think I read somewhere else that the FINDA state isn't transmitted to the printer after a certain period of time.  It is still operational and the sensor input is being used by the MMU, but the printer eventually treats it as a "don't care" input.

Similar to this post above:
https://github.com/prusa3d/Prusa-Firmware/issues/1770#issuecomment-502396289
But I think it can stop transmitting even when there is no error condition at the MMU.

nimaim, 2020-05-23:
Was this fixed in the new firmware (3.9.0)?

andrepruitt, 2020-05-25:
Yes.
Thanks you.
Andre

JohnnyDeer, 2021-01-20:
@n8maher this issue should be fixed, can you confirm that?

closed 2023-09-26 by github-actions[bot]