prusa3d/Prusa-Firmware #1772: Hotend crash into bed if Z lifted during heating

Cifik, 2019-04-27:
Hello,
i just faced following issue:
I started the print, lifted Z (by long-pressing button) to check the nozzle during heating phase.
When hotend and bed were heated, bed levelling should start, but printer homed X and Y and forced Z down until the reset (effectivly destroying the bed under nozzle).
Printer: Prusa Original i3 MK2.5S
FW: 3.7.0

titoxd, 2019-05-04:
Possible duplicates:
#1248
#1298
#1419

wavexx, 2019-05-05:
@titoxd Did you manage to reproduce this on the MK3?

I tweaked the Z homing code for another fix recently (see #1680 and #1759), where I did this several times in a row and never had an issue.

Reading the reports, the minimal steps I would need to do is have either "Move Z" or "Live adjust Z" active before Z is being homed, but that definitely wasn't enough for me.

leptun, 2019-05-05:
It might be impossible to reproduce on the mk3 as the endstop code is different. Also on the mk3 you have crash detection on Z axis, so even if you home in the bed it detects and aborts

wavexx, 2019-05-05:
@leptun As far I understood it, the state of the lcd encoder is *not* reset when Z homing is done, so this could definitely apply to the MK3 as well (see #1298, #1419).

I didn't look at it in depth, but if that's using a stale variable taken from the current_position and never updated while the menu is active, you're going to have a large shift on the first encoder update in both move Z and live adjust when planner_set_position is called behind it's back.

titoxd, 2019-05-06:
@wavexx It originally happened to me on the MK3. I did not try to do it again as I do not have a backup steel sheet / nozzle in case it gets damaged.

wavexx, 2019-05-06:
On Mon, May 06 2019, titoxd wrote:
> @wavexx It originally happened to me on the MK3. I did not try to do
> it again as I do not have a backup steel sheet / nozzle in case it
> gets damaged.

To test it safely, and if I read the report right, this happens early
enough that you could do something along these lines:

- home the machine from the lcd
- place a metal washer on a kitchen sponge (oriented vertically)
  directly under the pinda (verify that the pinda triggers on the washer
  first). Use the "move z" from the lcd to place it close enough
- run the following gcode using a gcode sender:

  G4 S30
  G28 Z

This will give you 30 seconds to fiddle with the LCD before homing Z only.
If the PINDA hits the washer and keeps going you now have more time to
hit reset.

Crude, but well, I did worse ;)

wavexx, 2019-05-06:
On Mon, May 06 2019, Yuri D'Elia wrote:
> This will give you 30 seconds to fiddle with the LCD before homing Z only.
> If the PINDA hits the washer and keeps going you now have more time to
> hit reset.

And to clarify, do not heat the bed or nozzle.

I don't think bed heating has anything to do with it.

kgolger, 2019-06-15:
This does not solve the problem, but avoids z-axis collisions (see pictures).

[image]
[image]

wavexx, 2020-01-27:
I was looking at the old issues where I commented and tried to reproduce this one.
It looks like now we do not allow to move Z and/or adjust live Z while heating and/or mesh bed leveling.

The menus are dismissed just before entering, so my first idea about the encoder code using old values cannot apply anymore (you cannot move Z anymore).

3d-gussner, 2020-07-13:
@Cifik Is this still an issue? If it has been fixed in a newer firmware version, please close this issue.

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer