prusa3d/Prusa-Firmware #1775: [Question] Why 15°C for min temp?

tobymurray, 2019-04-28:
Is the 15°C threshold based on anything? E.g. temperature sensor specs or practical experience? 

I regularly hit the `Err: MINTEMP` in the room my printer sits in, as the nozzle temperature appears to rest ~5°C below ambient. I've otherwise had no issues with my thermistor, and I have thousands of hours of prints with it so I would be surprised if the sensor itself was faulty. My current workaround is to use a portable heater to preheat the printer until it gets above 15°C, then carry on. I've recently become fed up with this as the nozzle temperature dropping to 15°C has caused print failures due to cooldown after a jam has been detected or something.

I'm considering modifying the configuration in the firmware to up it to something that seems more reasonable to me from what I've observed experimentally, but I don't want to do it if the 15°C has some safety significance.

JustAnother1, 2019-04-30:
The safety significance is, as far as I know, to detect sensor failures. Such as short circuit and disconnected. So Check the temperatures that your settings create if the sensor is disconnected and if you short circuit the sensor. One value should trigger max temp error the other value should trigger min temp error. As long as that still works there should be no issue to adopt the values as long as these two hold true.

tobymurray, 2019-04-30:
I just tried it out - a short sets off the MAXTEMP alarm and displays 739°C. An open circuit shows 0°C (and the MINTEMP. Looks like I'll be lowering the threshold then!

nevercast, 2019-06-27:
Are you saying you can't start printing when the ambient temperature is under 15C? I'm running V3.7.1 firmware on my MK3S and the ambient was 7°C when I started printing today. I did run a preheat though, perhaps preheat ignores the min temp that you are talking about? When does MINTEMP occur?

tobymurray, 2020-10-11:
Addressed in https://github.com/prusa3d/Prusa-Firmware/pull/1855

closed 2020-10-11 by tobymurray