prusa3d/Prusa-Firmware #1777: Support the latest IR sensor for filament on Rambo 1.3 boards (MK2.5S)

boilerbots, 2019-04-29:
The latest parts files for the Extruder are designed for the new IR optical gate board instead of the laser reflector board. The menu doesn't have a configuration option to select which type of board you are using and the latest board causes the self check to fail.

Also it is not clear which pin the output of the sensor should connect to, the connector that came with my board uses PH2 (pin 84) where the previous board used SDA for the white wire.

michalxfanta, 2019-04-29:
For the MK2.5S with Rambo 1.3 exists different HEX file https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.7.0/FW_MK25S-RAMBo13a.zip with firmware, which supports new IR filament sensor.

boilerbots, 2019-04-29:
I thought that is what I just loaded, I will check.

boilerbots, 2019-04-30:
I loaded the MK2.5 "not S" firmware, my mistake.

closed 2019-04-30 by boilerbots