prusa3d/Prusa-Firmware #1779: Print fan issue from a value of 75

mrandre79, 2019-04-29:
From a value of 75 and up it sounds like a cyclic fan start boost.

jsilva-iii, 2019-04-29:
I notice this as well the fans speed hunts and is not constant

boilerbots, 2019-05-01:
I just bought my 2.5S from a friend and finished the upgrades and loaded the latest firmware and I experienced this same thing. I am also seeing Thermal Runaway and wondered if that fan is running to fast with higher fan speeds.

rennne, 2019-05-07:
Maybe related to:
[3.5.2-RC1 - Print fan PWM bursts #1516](https://github.com/prusa3d/Prusa-Firmware/issues/1516)

Prusa-Support, 2022-05-11:
Hello! Sorry for the very late response on this issue. If still relevant, the "cyclic fan boost" is related with the fact that the fans generate at low RPM so much noise on the signal line that it is hard to distinguish if it is spinning or not: to be able to get some "readable" values the fan is forced to spin up.

About the Thermal Runaway issue after the upgrade to 2.5S, if still relevant: since it occurred right after the upgrade, it was likely related with the orientation of the fan shroud, or perhaps a misshapen fan shroud. It is relatively easy to "over-tighten" the bolt holding the fan shroud which would cause it to be at an angle pointing towards the heater block, which would cause the temperature drop (cold air onto the heater block) and the runaway. 

I'll close the issue for now, @boilerbots in case there might be need to further review the thermal runaways, please let me know via email info@prusa3d.com mentioning me in the subject ("attn: Alessandro"). 

Alessandro Pantaleo
Prusa Research

closed 2022-05-11 by Prusa-Support