prusa3d/Prusa-Firmware #1782: Resetting MMU resets printer as well - MK2.5S +MMU2S

jsilva-iii, 2019-04-29:
I had an issue with a filament jam and the MMU selector ended up getting stuck. After I cleared the jam, was MMU unable to load the correct filament - I attempted to reset the MMU2 (pressing reset button on the side of board) and ended up resetting the printer as well - the print was lost.  If I reset the MMU2  by pressing the button -  the printer gets reset as well - is this behaviour expected on the mk2.5?  I'm running FW 3.7.0 and 1.0.5 for MMU

leptun, 2019-04-29:
Tl;dr this is expected behavior. 

Unfortunately on the Mk2.5 and on the Mk2.5S there weren’t enough pins on the board/mmu connector. The solution to this was implementing software reset where the printer sends a reset command to the mmu. On those printers the reset lines of the mcu and mmu are tied together. That means if you reset one of them, they both reset.

On the Mk3+ the mmu reset line is connected to an IO pin of the mcu. This way the mcu can reset the mmu, but not the other way around.

jsilva-iii, 2019-04-29:
Thanks for the info and prompt response - are there any plans to put a SW reset option on the printer (or is it already there ?) as I see another user has requested it too.

leptun, 2019-04-29:
The software reset is currently used on startup. Not sure if it is implemented through gcode from octoprint or pronterface . It is not in the menus for sure.

I don’t know when or if it will get implemented. I don’t work at Prusa.

jsilva-iii, 2019-04-29:
Thanks again for your help. I'll close this as someone else has requested a SW reset

closed 2019-04-29 by jsilva-iii