prusa3d/Prusa-Firmware #1787: Power Panic input initialize

GWdd, 2019-04-30:
It appears that a short in the Power Panic switch or wiring will not be detected at power on, leaving the circuit to turn off heaters, but not detected by the CPU; so everything else appears to function normally, including starting a print.

This condition would happen if the firmware reads and stores the state of the power panic pin, and accepts a low as normal. Subsequent tests will detect no change and not react to the failed condition.

wavexx, 2020-01-27:
I don't have the reference of the rambo at hand, but if that can be done I guess a better place to put this would be during self-test

wavexx, 2020-01-29:
Related: #2446

The FW *should* definitely check for this to avoid losing state during short repeated interruptions. We might need to do something on top of that to help the user detect that the wiring is wrong.
Skipping the test when the printed is uncalibrated and delay it to the first self-test is an option.

closed 2023-09-28 by github-actions[bot]