prusa3d/Prusa-Firmware #1788: New_XZY Ignores to obey configuration_prusa.h

ingybing, 2019-05-01:
Hi,

I'm trying to use the firmware but I've had to override the configuration since my cables were wired as inverted, so i flipped the direction in configuration_prusa.h. Also my x/y motor is 80 steps per mm instead of 100. This is fine for most of the operation however when it performs the XYZ Calibration after returning from the Z inital calibration it travels to the build plate but then ignore the direction inversion and moves away from the heated bed instead of down. So it appears the new_xzy isn't taking into account any updates from stock. This seems to be backed up by a second problem that after physically changing the wiring over it finds the first point fine but then tries to travel to the second point using 100 steps / mm instead of 80 and collides with the right had Z Axis.

I'm digging through some of the code to see if i can work out whats wrong but i'm new to it so wonder if someone else has hit it / knows where i can look or just include something that's missing etc.

cimoalpacino, 2019-05-01:
You'll be able have to modify quite a bit of code.
I also had pulleys like you and needed 80 steps, but the calibration would always fail. I ended up buying proper pulleys to have 100 steps. 
You'll have to modify the direction of the stepper motors in sm4.c and your calibration points in xyzcal.cpp.

You can compare my gir repository here where i made changes to work with my ramps board. 

https://github.com/cimoalpacino/Prusa-Firmware/tree/MK3_3.7.0_ARDUINO_RAMPS/Firmware

Panayiotis-git, 2019-05-01:
I hope this could help: https://github.com/prusa3d/Prusa-Firmware/pull/1263

ingybing, 2019-05-02:
Thanks 🙂. Like a pleb I was fixated on a software solution and never considered just changing the pulleys. All done calibrated and printing 😃. 

All this hard coding to a particular step / homing sensitivity etc could potentially bite them in the bum later late if they want to  change something.

closed 2019-05-02 by ingybing