prusa3d/Prusa-Firmware #1792: MK3 FW 3.7.0 print recovery at wrong height after power panic

ligniant, 2019-05-03:
I had a long power outage (~20min) while printing. The printer correctly detected the power panic and offered me a recovery. But after homing x and y, preheating and extruding some filament, the printer started to print some millimeters to high (extruding filament into the air). The printer went up the nozzle while power outage to get away from the print but after resuming I have not seen the printer doing any height correction to get back to the original height.

YoungR0se, 2019-05-06:
Hi ligniant, I had this exact issue myself four days ago.  Power panic - resume print and the printer starts printing in the air!  I power panicked the printer again, manually moved the Z axis so the head was the correct height above the print, resumed print and the printer tried to lower the head, resulting in it crashing into the print.  It raised the head and did a XY home, during which time I manually screwed the Z axis up so the head was approximately where it had been after the first power panic, and the printer then lowered the head and began printing like normal (although I had to make some fine height adjustments involving calipers).  I have no idea why this happened, or why the second time the power panic function started printing at the correct height.

ligniant, 2019-05-06:
Hi YoungR0se, I have not tried to deliberately power panic the printer yet. But I think it could be correlated to firmware update. It was the first power panic for the printer since it has been built and so it was the first power panic with the new 3.7.0 firmware since flashing. (It is only an idea, because I have seen a similar behavior with the new 7x7 bed leveling, the first time after flashing the 3.7.0 firmware the first layer came out ugly, but from than on the first layer became perfect on the complete printing area, much better than ever before!)

wavexx, 2020-01-27:
This should be fixed with #2437 which just got merged.

Prusa-Support, 2022-06-04:
This issue can be closed because fixed in FW [v3.9.0-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0-RC1).

closed 2022-06-04 by Prusa-Support

BFOLogitech, 2023-09-01:
I just had this happen on an MK3S+ with the latest firmware (3.13.0). We had a power cut, the printer did it's thing and when it restarted when the power cam back on, it automatically started again but about 2-3cm above the last printed layer. Further to the comments above, this is the first power cycle since the firmware was updated yesterday