prusa3d/Prusa-Firmware #1793: Garbage on display

waddleducky, 2019-05-03:
Hello all, 

I am having a problem with the display. I flashed the new .hex file to my printer and it comes in amazing and fine. I am using a PT100 amplifier board and PT100 thermistor, which I need to flash the firmware through Arduino. The problem is when I do, the display is completely messed up. Just random symbols everywhere, and if I click it changes to more random symbols as well as scrolling but it will always end up at the same weird symbols. Is flashing the new 3.7.0 firmware btw. So I went and flashed the old 3.1.0 firmware and it goes through and everything works perfectly. Not sure why the LCD is messing up on the new firmware and why no one else is having the issue. 

Thanks!

Panayiotis-git, 2019-05-03:
Check the paragraph 2c at the compile instructions:

c. In file "Firmware/config.h" set LANG_MODE to 0.

IIRC, this will do the trick.

waddleducky, 2019-05-04:
> Check the paragraph 2c at the compile instructions:
> c. In file "Firmware/config.h" set LANG_MODE to 0.
> IIRC, this will do the trick.

You are amazing. I literally thought I did this first thing... apparently I didn't save it.

closed 2019-05-04 by waddleducky