prusa3d/Prusa-Firmware #1794: MK3S + MMU2S - 1st layer calibration does not pull filament

miguelmmm, 2019-05-04:
I recently updated my MK3+MMU2 to MK3S+MMU2.
Drivers updated to 3.7.0-2201 / 1.0.5-297
When I try to perform the 1st layer calibration the MMU2 does not push the filament all the way to the hot end. The pulley motor does a single forward revolution and then it rotates backwards for less than 2 revolutions, then hot end continues to execute the 1st layer calibration routine as if the filament reached the hot end (i.e. it doesn't consider the IR sensor).
During regular prints, the MMU has pushed the filament all the way to the extruder so I believe this is related to the logic/firmware, no mechanical issue at all. I can reproduce the issue consistently.
Here is a link to the video: https://youtu.be/R6PHi9YvCe0

Auroch, 2019-05-20:
Same for me here with MK3S+MMU2S
Firmware : 3.7.1-2266 / 1.0.5-297
The MMU push the filament to the extruder, then remove it sompletely as it would change filament,
then the extruder try to print but the filament is no more there ;-(

Stefan-Z-Camilleri, 2019-05-20:
Same here +1 (i.e. same problem, running 3.7.1-2266 and 1.0.5-297 as @Auroch )

First layer calibration is unpredictable... it worked just fine for about 3 or 4 runs.. then as OP describes:
1- Loads filament to extruder
2- Retracts it
3- Loads it again to extruder
4- Retracts it
5- Extruder moves to the center and does 'something'... but the movement pattern is different to the test pattern.

Following this, 'Calibration' option is no longer present in the menu.  Printer needs to be reset.

Stefan-Z-Camilleri, 2019-05-20:
Hey @Auroch / @miguelmmm , I've actually resolved this issue.  The problem was a miscalibration of the IR sensor.  You may want to check that.

mdwelsh, 2019-06-09:
Same problem here. Running first layer calibration causes filament to be loaded to the hot end, then retracted back to the MMU. This will sometimes repeat. It then tries to perform the first layer calibration with no filament loaded.

I checked that my IR sensor is working correctly following the instructions here:

https://help.prusa3d.com/article/wprwemqsmj-ir-filament-sensor-calibration

JordanGrantF15E, 2019-06-26:
I'm just getting my MMU installed and I have the same issue going on.

Auroch, 2019-06-27:
As said by Stefan, it was a miscalibration of the IR sensor for me.
But in my case, the calibration must be reapplied regularly despite a good tightening of the screws. It seems that there is a slight displacement that occurs due to the force of the printing movement.

JordanGrantF15E, 2019-06-27:
I think I figured it out. The filament sensor was set to "off". That may have been a carry over from the previous settings before I upgraded the firmware. After turning the filament sensor setting to "on", the first layer calibration routine seemed to start working the way I would expect.

eiannone, 2019-07-12:
Thank you very much @JordanGrantF15E, I was in the same situation, and the problem was fixed just by re-enabling the filament sensor in settings.

JordanGrantF15E, 2019-07-12:
Great! Glad I could help.

JohnnyDeer, 2021-01-20:
This is not the FW issue as was mentioned here this behavior is caused by IR tuned off. I´m closing.

closed 2021-01-20 by JohnnyDeer